jgromes/RadioLib #339: CC1101_Receive does not work on ESP8266; Interrupt issue?

xstff78, 2021-07-29:
Hi,
First of all, thank you for creating and maintaining this great, super versatile library.

I cannot get the CC1101_Receive example sketch working as expected on an ESP8266; please find the procedure and the error description below:

I connected two CC1101 modules to ESP8266 boards (two WeMos D1 mini Lite, later exchanged by two NodeMCU 1.0) and used the RadioLib (_v 4.5.0_) examples via PlatformIO (Espressif 8266 version _3.1.0_).

Beside GND and 3.3 V, the following connections were made:
D1--DGO0, D2--GDO2, D5--SCK, D6--MISO, D7--MOSI, D8—CSN.
Therefore, I changed the corresponding line in the CC1101_Transmit (and _Receive) example to 

<code>CC1101 radio = new Module(D8, D1, RADIOLIB_NC, D2);</code>

The transmit example works as expected on both units (tested by an SDR and URH).

The receive sketch initialised successfully, but did not receive anything (“failed, code -6” which is the ERR_RX_TIMEOUT).

After trying different changes without success (NodeMCU instead of WeMos boards, baudrate, output power, distance between tx and rx), I hooked up an oscilloscope and found that there were regular low pulses on the GDO0 line, indicating successful receipt of the packets.

To me it seems that the interrupt triggering the readout of the CC1101 does not work correctly.

I tried to dive into the code to find the cause of this behaviour, but did not succeed due to my beginner skills.
I hope that someone can reproduce the findings and help to solve them in the next versions.

Kind regards,
Christian

jgromes, 2021-07-29:
First of all, CC1101_Receive is an example of the blocking `receive()` method. If you don't get a packet during the reception window, you will get ERR_RX_TIMEOUT. Try C1101_Receive_Interrupt for an interrupt-driven version.

Also, does the timing of the pulses correspond to the timing of packet transmissions?

xstff78, 2021-07-29:
Thank you for your quick reply.

You are right; I overlooked the fact that only the CC1101_Receive_Interrupt example uses interrupts obviously.
And it works fine with my setup, I just had to change the ISR line to <code>IRAM_ATTR void setFlag(void) {</code> due to the ESP8266.

I wonder why I am not able to receive any data with the nomal Receive example. In order to send data within the reception window, I lowered the baud rate and also tried to decrease the delay in the Transmit example to 1 ms, but without any success.

The low pulses on the GDO0 definitely indicate a valid data packet – the loop delay in Transmit corresponds to the interval of the pulses.

jgromes, 2021-07-29:
I'm not sure why the blocking receive wasn't working on your setup, since internally it's calling the same methods as the interrupt example.

Closing now as the issue seems to have been resolved.

closed 2021-07-29 by jgromes