jgromes/RadioLib #340: SX1278 initializing failed: code -16

mputnieks, 2021-07-30:
I understand that error code -16 refers to ERR_SPI_WRITE_FAILED. However, I have checked and all the SPI connections are correct. Would this suggest my module is broken and if so, is there any other way to check for that using the RadioLib? Thank you in advance!

jgromes, 2021-07-30:
I suggest starting by filling out the "Module not working" issue template, as key information such as which Arduino you're using, or what deos your hardware setup look like are missing.

> is there any other way to check for that using the RadioLib

Of course, there's the [debug mode](https://github.com/jgromes/RadioLib/wiki/Debug-mode)

mputnieks, 2021-07-30:
Thank you for the quick reply! As per the template:

**Sketch that is causing the module fail**

#include <Arduino.h>

/*
   RadioLib SX127x Transmit Example

   This example transmits packets using SX1278 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX127x/RFM9x family can also be used.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx127xrfm9x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   D7
// DIO0 pin:  D5
// RESET pin: D8
// DIO1 pin:  D6
SX1278 radio = new Module(7, 5, 8, 6);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */
}

void loop() {
  Serial.print(F("[SX1278] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX127x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}

**Hardware setup**

//   NSS pin:      D7
//   DIO0 pin:    D5
//   RESET pin:   D8
//   DIO1 pin:    D6
//   MISO pin:   D12
//   MOSI pin:   D11
//   SCK pin:      D13

**Debug mode output**
[SX1278] Initializing ... R     42      12
M       SX127x   
R       1       9
R       1       9
W       1       9
R       1       9
R       24      5
W       24      0
R       24      0
R       1       9
R       1       9
R       1       9
W       1       8
R       1       9

address:        0x1
bits:           2 0
value:          0b0
current:        0b1001
mask:           0b111
new:            0b1000
read:           0b1001

R       1       9
W       1       89
R       1       9

address:        0x1
bits:           7 7
value:          0b10000000
current:        0b1001
mask:           0b10000000
new:            0b10001001
read:           0b1001

R       1       9
R       1       9
W       1       9
R       1       9
failed, code -16

**Additional info (please complete):**
 - MCU: Arduino Nano ATmega328 (New Bootloader)
 - Wireless module type : sx1278
 - IDE: VS Code, Platformio add-on
 - Library version : 4.5.0

jgromes, 2021-07-30:
You're using a 5V Arduino Nano, do you have a logic level shifter between that and the 3.3V SX1278?

mputnieks, 2021-07-30:
I am using a USB connected to my pc to power the setup, not a battery, hence it should work as far as I am aware. The SX1278 is, of course, connected to the 3.3V pin.

jgromes, 2021-07-31:
1. It (almost) doesn't matter what is powering the Arduino - it has an on-board voltage regulator (usually some 1117), which has a limited output current, but that is completely unrelated because:
2. That's the power pin, what are you using to convert the 5 V MOSI, NSS and SCK signals from Arduino into 3.3V signals for the SX1278? It has a 3.3V logic, so 5V signals coming from your Arduino will damage it.

mputnieks, 2021-07-31:
Whoops, I am using exactly nothing. I hope it still works once I get the level shifter set up in between. Completely missed this myself, thank you so much for the support!

jgromes, 2021-07-31:
It is possible you have damaged your module. From the debug log, it seems to be unable to enter sleep mode, which would be very strange under normal circumstances. I also seem to recall similar behavior with SX1278 that were connected to 5V Arduino boards.

Closing for now, as the issue seems hardware/usage-related.

closed 2021-07-31 by jgromes

mputnieks, 2021-08-03:
Hi again, another related question: I was wondering, do MISO, DIO0, DIO1 and RESET pins also need to be passed through the logic level converter?

jgromes, 2021-08-03:
They sohouldn't unless your logic level shifter can handle simultaneous bidirectional signals. It's easier to just connect them directly to Arduino, it will safely detect the 3.3 V level as logic high signals.

mputnieks, 2021-08-12:
Hi there, so I ordered and installed voltage regulator and logic level converter. Now I'm wondering whether I need to order a new lora too, as this one might be damaged...

Sadly, I got the following error output: (same code as before)

```
[SX1278] Initializing ... R     42      FF
SX127x not found! (1 of 10 tries) SX127X_REG_VERSION == 0x00FF, expected 0x0012
R       42      FF
SX127x not found! (2 of 10 tries) SX127X_REG_VERSION == 0x00FF, expected 0x0012
R       42      FF
SX127x not found! (3 of 10 tries) SX127X_REG_VERSION == 0x00FF, expected 0x0012
R       42      FF
SX127x not found! (4 of 10 tries) SX127X_REG_VERSION == 0x00FF, expected 0x0012
R       42      FF
SX127x not found! (5 of 10 tries) SX127X_REG_VERSION == 0x00FF, expected 0x0012
R       42      FF
SX127x not found! (6 of 10 tries) SX127X_REG_VERSION == 0x00FF, expected 0x0012
R       42      FF
SX127x not found! (7 of 10 tries) SX127X_REG_VERSION == 0x00FF, expected 0x0012
R       42      FF
SX127x not found! (8 of 10 tries) SX127X_REG_VERSION == 0x00FF, expected 0x0012
R       42      FF
SX127x not found! (9 of 10 tries) SX127X_REG_VERSION == 0x00FF, expected 0x0012
R       42      FF
SX127x not found! (10 of 10 tries) SX127X_REG_VERSION == 0x00FF, expected 0x0012
No SX127x found!
failed, code -2
```
Does this indeed point to a damaged module or some other issue I might be missing? Thank you again!

jgromes, 2021-08-12:
Your module is outputting 0xFF (all 1s), which can be commonly seen in case of disconnected/incorrect wiring (you can try it - disconnect everything from the Arduino, the output will beeither 0xFF or 0x00).

I'd suggest checking the wiring. Which level shifter are you using?

mputnieks, 2021-08-13:
Removed all extra modules used in the project, rewired the lora, voltage regulator and logic level shifter(ICQUANZX 4 Ch. IIC I2C) using connection-checked wires, and lo and behold, it works!

jgromes, 2021-08-14:
I'm glad it's working now, however, are you sure you're using a logic level shifter? Google seems to think ICQUANZX 4 Ch. IIC I2C is an A/D converter.

mputnieks, 2021-08-15:
This is the exact one I ordered: [https://www.amazon.co.uk/dp/B07RDHR315/ref=pe_27063361_487055811_TE_dp_2](url)
pretty sure it's not an A/D converter, also no way it would work if it were

jgromes, 2021-08-15:
Ah, so it's just some BSS138 MOSFETs.

> pretty sure it's not an A/D converter, also no way it would work if it were

My thoughts exactly, that's why I was asking :)