jgromes/RadioLib #345: CC1101 + ESP32 receiving random junk data

0x5c4r3, 2021-08-04:
Just reporting the bug I explained in the Discussion, hoping for someone to solve the issue or give me directions to solve it.

**Description of the bug**
Using the default script `CC1101_Receive_Interrupt` with a C1101 and an ESP32, the receiver will dump junk data and won't be able to detect any other signal. The receiver is tuned at 433.92MHz.

The junk data looks like the following:
```
Control - Received packet!
[CC1101] Data:		$ H�S⸮⸮⸮⸮⸮⸮⸮gݧ⸮⸮:W⸮�{⸮N⸮ԇ⸮�⸮⸮,⸮⸮�⸮⸮��⸮[:~p⸮⸮֋�Qh;x1⸮�͖_p⸮⸮X⸮<a⸮̚doD⸮⸮P⸮⸮⸮⸮d⸮⸮o+⸮⸮�⸮⸮⸮ۚ!	⸮⸮�⸮Q�⸮⸮⸮⸮en=⸮⸮0⸮⸮⸮⸮C�⸮Nc⸮
h$6A⸮⸮hA?⸮⸮⸮M⸮P�⸮�⸮�⸮'�⸮J]C⸮⸮)⸮⸮PG⸮⸮�⸮⸮g⸮81�⸮⸮⸮⸮�⸮�⸮⸮⸮6�⸮?⸮⸮W⸮ '⸮a⸮⸮ؓ⸮⸮m⸮B⸮⸮ Ƭ8⸮⸮cP⸮Q⸮x⸮⸮�E⸮
Control - Received packet!
[CC1101] Data:		⸮⸮⸮⸮⸮⸮⸮⸮⸮�⸮⸮#)⸮Q�⸮�⸮⸮p⸮jDT�⸮�3JO)
⸮⸮⸮˒[⸮#⸮⸮I�⸮⸮⸮⸮,X⸮⸮(⸮r@⸮*߸dw⸮!⸮⸮⸮}n�⸮�⸮⸮DG⸮⸮�⸮2`46f⸮!⸮!⸮y⸮ *⸮k⸮⸮⸮�⸮	⸮!)*⸮�⸮"⸮S⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮]⸮��⸮⸮dm⸮⸮⸮⸮0us⸮⸮M)�⸮⸮�H�e⸮⸮⸮!ow⸮⸮⸮(.�'⸮�� �W⸮��⸮
CRC error!
Control - Received packet!
[CC1101] Data:		⸮@4�A�
CRC error!
Control - Received packet!
[CC1101] Data:		⸮⸮a$t⸮k⸮⸮⸮!⸮⸮⸮�⸮��$OB⸮⸮?K3⸮��:⸮��⸮?⸮B⸮⸮v?⸮
⸮M⸮�5⸮⸮⸮G�⸮⸮fA⸮⸮⸮⸮dG⸮��⸮z⸮
Control - Received packet!
[CC1101] Data:		 ⸮�⸮⸮�>K⸮�⸮�⸮S�⸮⸮⸮⸮�⸮�&⸮H�t�֊⸮⸮�⸮⸮\>⸮]�m/u⸮⸮�⸮�⸮�⸮⸮J⸮0{���3⸮⸮��⸮f['	⸮`�qS��+����⸮�5K⸮⸮⸮5⸮ҰY⸮⸮a⸮�⸮⸮⸮�8*
⸮w⸮~�M⸮{
Control - Received packet!
[CC1101] Data:		⸮0-⸮⸮!⸮!⸮�⸮⸮⸮⸮`⸮~B ⸮
⸮⸮⸮y⸮�⸮"c⸮n1�⸮6⸮�#⸮˩5⸮⸮⸮9⸮⸮j⸮Cd⸮
a⸮C=A֧⸮5⸮<E⸮⸮V⸮⸮��⸮⸮⸮�⸮⸮⸮qT⸮s⸮>P⸮U⸮⸮���u⸮� ⸮q�⸮⸮⸮⸮91⸮�AB⸮I⸮⸮�⸮1�⸮A⸮Qe⸮⸮⸮⸮V��⸮⸮ۑ K⸮b⸮1�'⸮⸮⸮�Ac⸮� \�⸮⸮⸮;⸮&#⸮݅⸮
Control - Received packet!
[CC1101] Data:		⸮8⸮ρ	⸮⸮⸮�ө⸮⸮P�⸮G⸮⸮�⸮�⸮I⸮���
CRC error!
Control - Received packet!
[CC1101] Data:		�u�
CRC error!
Control - Received packet!
[CC1101] Data:		⸮�⸮jm⸮1g⸮Z⸮A⸮⸮⸮⸮i⸮⸮̑⸮
```

If trying to use the default `CC1101_Receive`, the receiver will not detect any signal from the transmitter, and will keep outputting `error -6`.

**Expected behavior**
Instead of dumping random data, the receiver should be able to dump the package sent from another transmitter (I tried several, none of them is detected).

**Additional info (please complete):**
 - Board used: ESP-wroom-32
 - Wireless module type: CC1101
 - Arduino IDE 1.8.15

**Additional Checks**
- added an LM2596 + powerbank to power the board. The issue is proved not to be related to power since it's still there.
- changed pins/jumpers multiple times (direct connections, breadboard not used), the issue is proved not to be related to wiring since it's still there.
- IRAM used.
- Code double checked.

jgromes, 2021-08-19:
@Guglio95 I know you guys use CC1101 quite a bit, have you noticed this behavior? It's not related to ESP32 being used, it happens on Arduino Uno as well.

Guglio95, 2021-08-21:
Hi @jgromes ,
We use the transceiver in “promiscuous” mode and it is common for us to get garbage-packets.

Anyway, when I first tried RadioLib with a transmitter and a receiver I think I wasn’t receiving  random data.

jgromes, 2022-10-09:
I recently tried reproducing this issue, but I was unable to do so. While there still are random packet being received, they all return CRC error now, so I'm not sure if that is the original issue or whether it was resolved. Anyway, if anyone has a similar issue with the newer versions of the library, feel free to open a new issue.

closed 2022-10-09 by jgromes