jgromes/RadioLib #346: AX25 AFSK tones

lw2dtz, 2021-08-04:
Hi

I started to test the new version of the library with the following scheckt of AX25_AFSK. I am using an sx1278 and a STM32DUINO adapted from a blue pill. Although it generates the AX25 frame, the tones are frequency shifted downwards as shown in the photo, although they are decodable.

Regards.

Gustavo.

[AX25_Transmit_AFSK.txt](https://github.com/jgromes/RadioLib/files/6933963/AX25_Transmit_AFSK.txt)
[registers.txt](https://github.com/jgromes/RadioLib/files/6933965/registers.txt)
[image]

 - MCU: [ STM32F103F3C]
 - Link to Arduino core: [STM32DUINO]

 - Arduino IDE version  [e.g. 1.8.13]
 - Library version [4.45]

jgromes, 2021-08-07:
This could be due to a couple of factors - first of all, you should be demodulating as FM, could you check?.

Next, the tones are generated by your Arduino's tone() function, RadioLib only does the inital configuration of the module, so it should have very little impact on the frequency of the tones.

Testing with an Arduino Uno and SX1278, the tones are at roughly 1100/2300 Hz, so they're off by about 100 Hz, though centered correctly.

lw2dtz, 2021-08-08:
To your question, yes I using FM demodulation.

Identify the variables in your library, the value of the tones and change their value to work with the STM32, 
and they were like this:

#define AX25_AFSK_MARK                               1800
#define AX25_AFSK_SPACE                               2800

I was able to decode the frames successfully with the UZ7HO and MIXW software.

It remains the work of modifying the library to adjust the tones according to the micro
that is used. I hope to try with an arduino-zero.

jgromes, 2021-09-05:
@lw2dtz I added a `setCorrection` method to set offsets for mark and space frequencies to help deal with this. Unfortunately W wasn't able to dicsover the root cause.

BTW what is your SDR setup? I wasn't able to see as well defined mark and space frequencies in the soundmodem waterfall as you ahve on the screenshot you posted.

closed 2021-09-05 by jgromes