jgromes/RadioLib #347: Odd problem when works with other LoRa library.

xg590, 2021-08-11:
I am using a MicroPython LoRa library (Raspberry Pi Pico + Adafruit RFM95W (SX1276) ) to receive message from RadioLib (ESP32 with on-board SX1276).

The problem is I can only get 4 characters ("WWII" or "1945" or "look") reliably and more characters sent from RadioLib are either random characters or garbage. 

If two tested LoRa systems both run RadioLib or both run the MicroPython library, everything works perfectly. 

**To Reproduce** 
```
#include <RadioLib.h> 
SX1276 radio = new Module(RADIO_CS_PIN, RADIO_DI0_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN);
void setup()
{
    initBoard(); delay(1500); 
                        //   freq,    bw, sf,  cr,          syncWord,  power, preambleLength,   gain
    int state = radio.begin(  915, 125.0,  12,  8,  SX127X_SYNC_WORD,      4,              8,     0); 
}
 
void loop()
{    
    int state = radio.transmit("1956abc");   
    delay(5000);
}
```
MicroPython
```python
import time, urandom
from machine import SPI, Pin 
####################
#                  #
#     1.Enable     #
#                  #
####################
# RFM95W         Pico GPIO
LoRa_MISO_Pin  = 16
LoRa_CS_Pin    = 17
LoRa_SCK_Pin   = 18
LoRa_MOSI_Pin  = 19
LoRa_G0_Pin    = 20 # DIO0_Pin
LoRa_RST_Pin   = 21
LoRa_EN_Pin    = 22

# Enable LoRa Module
lora_en_pin = Pin(LoRa_EN_Pin, Pin.OUT)
lora_en_pin.on()
time.sleep(0.01)

# Reset LoRa Module
lora_rst_pin = Pin(LoRa_RST_Pin, Pin.OUT)
lora_rst_pin.off()
time.sleep(0.01)
lora_rst_pin.on()
time.sleep(0.01)

####################
#                  #
#      2.SPI       #
#                  #
####################
'''
We command LoRa module to perform Tx/Rx operations via the SPI interface.
We disable SPI communication first to ensure it only happends when we need.
Define communication functions read and write.
The SPI comm is enabled temporarily for reading and writing and disabled thereafter.
'''
# Disable SPI communication with the LoRa module
lora_cs_pin  = Pin(LoRa_CS_Pin, Pin.OUT)
lora_cs_pin.on() # Release board from SPI Bus by bringing it into high impedance status.
time.sleep(0.01)

# SPI communication
[... 186 lines cut ...]
How do we know which area is for Tx and which is for Rx.
We must set the base addresses RegFifoTxBaseAddr and RegFifoRxBaseAddr independently.
Since SX1276 work in a half-duplex manner, we better set both base addresses
at the bottom (0x00) of the FIFO buffer so that we can buffer 256 byte data
during transmition or reception.
'''
RegFifoTxBaseAddr = 0x0e
RegFifoRxBaseAddr = 0x0f
Fifo_Bottom       = 0x00 # We choose this value to max buffer we can write (then send out)
write(RegFifoTxBaseAddr, Fifo_Bottom)
write(RegFifoRxBaseAddr, Fifo_Bottom)

write(RegOpMode, Mode['STANDBY'])  # Request Standby mode so SX1276 performs reception initialization.   
# Receive Data
write(RegOpMode, Mode['RXCONTINUOUS'])          # Request Standby mode so SX1276 send out payload  
```

**Expected behavior**
I expect 1956abc but get 1956\x16pI

**Additional info:**
Arduino core: [ESP32 1.0.6]
Arduino IDE version [1.8.15]
RadioLib version [4.5.0]
MicroPython: rp2-pico-20210618-v1.16.uf2

jgromes, 2021-08-12:
From the description it looks like some configuration inconsistency between RadioLib and uPython. What happens if the direction of the transmission is reversed, i.e. uPython -> RadioLib?

xg590, 2021-08-13:
I just prepared four systems: Two ESP32 with RadioLib and Two Raspberry Pi Pico with MicroPython. 
When they run simultaneously, each pair can communicate internally. Only MicroPython Code responds to RadioLib code (ValidHead) whereas RadioLib is tone-deaf to MicroPython code. 

* Can you explain to me how RadioLib handles packet header? Is header explicit or implicit? 
* Which packet-related parameters were set by RadioLib for SX1276? Bandwidth, coding rate or so?
* Is FHSS to be blamed? Does RadioLib use FHSS?

jgromes, 2021-08-13:
To answer your questions:

1. For SX1276 in LoRa mode, explicit header is used by default. There's also a header integrity check:  

https://github.com/jgromes/RadioLib/blob/2ccebd1fdc52f1858f64d93894a78ecad84978af/src/modules/SX127x/SX127x.cpp#L509-L515

2. See the [default configuration Wiki page](https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx127xrfm9x---lora-modem).

3. RadioLib doesn't use frequency hopping - it's not implemented and disabled by default.

> RadioLib is tone-deaf to MicroPython code

What does it mean exactly? Does it not receive the transmission at all? Does it return some error code? Does it return mangled data ...?

xg590, 2021-08-13:
When I enable RxPayloadCrcOn Tx side, RadioLib always throws out ERR_CRC_MISMATCH. When I disable it, RadioLIb sometime does print the message. Msg again begins with four and only four correct characters. And other time, I still get ERR_CRC_MISMATCH. 
What kind of mis-configuration could lead to this weird problem.

jgromes, 2021-08-14:
I was taking a look at the Python code, I noticed this:

> write(RegModemConfig3, 0x04) # 0x04 is SymbTimeout(7:0)

You don't seem to be enabling low data optimization. As per SX1276 datasheet recommendation, RadioLib enables this automatically for LoRa settings that produce symbols longer than 16 ms. Your configuration SF12/BW 125 kHz will produce 32.77 ms symbols, so you should enable that on the uPython side.

Alternatively, you can disable it on RadioLib side by calling `forceLDRO(false)`

xg590, 2021-08-14:
[Thumbs up] Thanks for you magical tip :) It works after I enable LowDataRateOptimize! 
From 
```
        write(RegModemConfig3, 0x04) 
```
```
        LowDataRateOptimize = {'Disabled':0b0, 'Enabled':0b1}
        AgcAutoOn = {'register LnaGain':0b0, 'internal AGC loop':0b1}
        write(RegModemConfig3, LowDataRateOptimize['Enabled'] << 3 | AgcAutoOn['internal AGC loop']<< 2)  
```

closed 2021-08-14 by xg590

jgromes, 2021-08-14:
Glad it's working now!