jgromes/RadioLib #351: SX1278 auto ACK

szerwi, 2021-08-24:
Hi,

Is there any way to use auto ACK with SX1278 (RF95 / Ra-02) module in this library? I would like to know whether the message was delivered successfuly or not.
If it's not possible in this library, is there any other library that has this feature?

jgromes, 2021-08-24:
As far as I know, SX127x has no built-in "auto-ACK" feature, similar to e.g. nRF24. Since RadioLib is primarily a collection of drivers, if it's not a feature of the module, you will have to implement it yourself in the application level - although I suspect implementing ACK mechanism is probably not all that difficult.

szerwi, 2021-08-24:
@jgromes thank you for the answer. Are there any additional modules for the RadioLib libraries the same as for RadioHead library? RadioHead has a RHReliableDatagram for addresable and acknowledged transmission or RFMesh for creating the mesh network. Is any of these possible using the RadioLib?

jgromes, 2021-08-24:
As far as I know there's nothing directly equivalent to RHReliableDatagram/RFMesh. The design intent for RadioLib was not to re-do work done in RadioHead, but rather to provide a simpler, yet more powerful interface to various radio modules.

Since this is not an issue, I'm converting it to a discussion instead.

closed 2021-08-24 by jgromes