jgromes/RadioLib #354: nRF24 Initializing Failed with -16

steven52880, 2021-08-29:
I have another problem today. My NRF24L01 module can not work. Could you please help me？

**Additional info (please complete):**
 - MCU: ESP8266
 - Wireless module type: NRF24L01(BK2425)
 - Arduino IDE version: 1.8.10
 - Library version: 4.5.0

**Sketch that is causing the module fail**
nRF24 Transmit Example

```c++
// include the library
#include <RadioLib.h>

// nRF24 has the following connections:
// CS pin:    15
// IRQ pin:   4
// CE pin:    5
nRF24 radio = new Module(15, 4, 5);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//nRF24 radio = RadioShield.ModuleA;
```

**Hardware setup**
```
NRF24	NodeMCU	8266
VCC	3V3	3V3
GND	GND	GND

CSN	D8	IO15
CE	D1	IO5
MOSI	D7	IO13
SCK	D5	IO14
IRQ	D2	IO4
MISO	D6	IO12
```

[image]

**Debug mode output**
```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 01a0
3fffff80:  fffffff0 3ffee308 3ffee458 402010bc  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  3fffdad0 00000000 3ffee480 402034b4  
3fffffb0:  feefeffe feefeffe 3ffe84e0 40100ff9  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
[nRF24] Initializing ... R	3	3	
M	nRF24
R	0	C	
W	0	C	
R	0	C	
R	4	55	
W	4	55	
R	4	55	
R	1D	0	
W	1D	4	
R	1D	0	
R	1D	0	
R	1D	0	
R	1D	0	
R	1D	0	
R	1D	0	
R	1D	0	
R	1D	0	

address:	0x1D
bits:		2 0
value:		0b100
current:	0b0
mask:		0b111
new:		0b100
read:		0b0

failed, code -16

```

jgromes, 2021-08-29:
It seems like your module is unable to enable dynamic payload length - writing to one of the SPI registers seems to have no effect.

You could try to ifnore the error by commenting out the following line, however this seems quite strange.
https://github.com/jgromes/RadioLib/blob/304e876c02d46f9fd8db6b49ae0eee937358e453/src/modules/nRF24/nRF24.cpp#L544

steven52880, 2021-08-30:
I tried to comment line 544 and 548
[image]

But it still failed with -16

Here are the logs
```

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 01a0
3fffff80:  fffffff0 3ffee308 3ffee458 402010bc  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  3fffdad0 00000000 3ffee480 402034b4  
3fffffb0:  feefeffe feefeffe 3ffe84e0 40100ff9  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
[nRF24] Initializing ... R	3	3	
M	nRF24
R	0	7E	
W	0	7E	
R	0	7E	
R	4	55	
W	4	55	
R	4	55	
R	1D	0	
W	1D	4	
R	1D	0	
R	1D	0	
R	1D	0	
R	1D	0	
R	1D	0	
R	1D	0	
R	1D	0	
R	1D	0	

address:	0x1D
bits:		2 0
value:		0b100
current:	0b0
mask:		0b111
new:		0b100
read:		0b0

R	1C	0	
W	1C	3F	
R	1C	0	
R	1C	0	
R	1C	0	

address:	0x1C
bits:		5 0
value:		0b111111
current:	0b0
mask:		0b111111
new:		0b111111
read:		0b0

R	7	E	
W	7	7E	
R	7	E	
R	7	E	
R	7	E	

address:	0x7
bits:		6 4
value:		0b1110000
current:	0b1110
mask:		0b1110000
new:		0b1111110
read:		0b1110

R	0	7E	
W	0	7E	
R	0	7E	
R	7	E	
W	7	7E	
R	7	E	
R	7	E	
R	7	E	

address:	0x7
bits:		6 4
value:		0b1110000
current:	0b1110
mask:		0b1110000
new:		0b1111110
read:		0b1110

R	0	7E	
W	0	7E	
R	0	7E	
failed, code -16
```

jgromes, 2021-08-30:
It seems like none of the SPI writes have any effect on any of the registers. Looking at the photo, are you sure it's an nRF24 module? The chip itself is covered with a black epoxy blob, so there's no way to tell if there's actually nRF24 underneath it. If I recall correctly, I also had some issues with these modules, which only were resolved by replacing with them with genuine nRF24, on which the package marking was visible.

steven52880, 2021-08-30:
I read the module introduction carefully and I'm sure it it BK2425 inside.
I search the chip info on Google and find [another lib](https://github.com/jnk0le/RFM7x-lib). It says "ones of the worst of chinese nRF24l01+ alternatives" It seems I have bought someting wrong :( . But the online store said it can be used the same as nRF24.

jgromes, 2021-08-30:
Thank you for the link to that library, it contains a lot of very interesting information about nRF24 clones/fakes. Briefly skimming through it, it looks like it should be possible to get those modules working with RadioLib, but to be honest, I'm not entirely sure it's worth the trouble.

> But the online store said it can be used the same as nRF24.

I'd try to get a return if I were you, that is clearly false advertisement.

closed 2021-08-30 by jgromes