jgromes/RadioLib #367: SX1276 unable to receive packets

kuestess, 2021-09-21:
@jgromes Thanks for the excellent library!  I am using a Heltec Lora 32 (v2) in an attempt to receive Insteon packets.  As you're probably aware, the Heltec Lora v2 is simply an SX1276 integrated with an ESP32 - pinout for the module is here: [pinout](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/PinoutDiagram/WIFI_LoRa_32_V2.pdf).  Insteon RF packets are 2-FSK packets of variable length, details of the packet format are here: [Insteon packet information](https://github.com/merbanan/rtl_433/blob/master/src/devices/insteon.c).  

For now, I'm simply trying to receive packets and will work through the decoding later.  I'm using the sketch below based upon the receive_direct example (which I think is the right place to start), but am unable to receive any packets.  I've also tried sketches based upon the FSK modem example, but am unable to receive packets there as well.  I am using RadioLib 4.5.0.  Any help is very much appreciated!

```c++
#include <RadioLib.h>

SX1276 radio = new Module(18, 26, 14, 35);

// DIO2 pin: 34
const int pin = 34;

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with FSK modem at 9600 bps
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.beginFSK(914.95);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  uint8_t syncWord[] = {0xCE, 0xAA};
  state = radio.setSyncWord(syncWord, 2);
  
  // set function that will be called each time a bit is received
  radio.setDirectAction(readBit);

  // start direct mode reception
  radio.receiveDirect(); 
}

// this function is called when a new bit is received
IRAM_ATTR void readBit(void) {
  // read the data bit
   radio.readBit(pin);
}

void loop() {
  if(radio.available() >= 1) { //shortened to 1 for testing purposes
    Serial.println("[SX1278] Received packet in direct mode!");
    while(radio.available()) {
      // read a byte
      byte b = radio.read();
  
      // print it
      Serial.print(b, HEX);
      Serial.print('\t');
      Serial.write(b);
      Serial.println();
    }
  }
}
```

jgromes, 2021-09-21:
First of all, you're using `setSyncWord`, which sets sync word for packet operation. If you want to try direct mode, you will have to call `setDirectSyncWord`.

Also, are you sure you have things like frequency deviation and bit rate configured properly? You don't seem to be doing that in the `beginFSK` method.

jgromes, 2021-09-21:
Also this seems more like a usage question than an actual issue, so until there's some problem in the library I'm converting this to discussion instead.