jgromes/RadioLib #368: Cant initialize NRF24L01 with autoack off

derungebundener, 2021-09-21:
Hi. i have a little problem:

I want to set custom settings to my NRF - channel (fr), data rate and power.

All works good:

  Serial.print(F("[nRF24] Initializing ... "));
  int state = radio.begin(2525, 250, 0, 5);
radio.setCrcFiltering(true);
radio.setAutoAck(true);

But i cant to turn off auto ack - with "false" i can see only "initializing".

Any idea?

jgromes, 2021-09-21:
Let's address the simpler stuff first:

> And if i understand right - all thiese state-codes we need only for debugging, yes?

No, the status codes serve a very important purpose - to let you know if what you think happened with the radio did actually happen. For example, since you ignore the return value of `setCrcFiltering`, you have no idea whether it was successfully enabled or not. There could have been a communication issue on the SPI bus that prevented the command to get processed. So ignoring status codes is never a good idea.

Now to the bigger problem: you have provided very little information as to what is the actual issue. What's your platform? What's the hardware setup? What is the output of debug mode? Etc.

You may have noticed we have issue templates, including a [Module not working](https://github.com/jgromes/RadioLib/blob/master/.github/ISSUE_TEMPLATE/module-not-working.md) template. If you follow it and include all the information it asks for, not only will it help me help you solve the issue, it will actually point you in the correct direction to debug it yourself.

derungebundener, 2021-09-22:
I using arduino nano (clone), and modified wiring:

// CS pin:    9
// IRQ pin:   10
// CE pin:    8
nRF24 radio = new Module(9, 10, 8);

Exmaple with simple (default setting) works good:
int state = radio.begin ();

With my settings also woorks good:
int state = radio.begin (2525, 250, 0, 5);

With CRC and ACK are both true agaiin good.

CRC is false and ACK is true- good.

But with ACK is false:
int state = radio.begin (2525, 250, 0, 5);
radio.setCrcFiltering (false);
radio.setAutoAck (false);

it writes only one first string - "Initializing ...", that means module can not execute "begin"

jgromes, 2021-09-22:
Unfortunately this issue is still missing a lot of the information that I need, namely the output of the [debug mode](https://github.com/jgromes/RadioLib/wiki/Debug-mode). This is also stated in the [Module not working](https://github.com/jgromes/RadioLib/blob/master/.github/ISSUE_TEMPLATE/module-not-working.md) template, which you still seem to have not read.

Also - you can't use pin 10 as IRQ, only pins 2 and 3. I suggest on reading up on interrupts on Arduino.

derungebundener, 2021-09-22:
Oh, okay thanks - dint know about 2 and 10, i was sure IRQ is usual GPIO pin.

jgromes, 2021-09-27:
I tried this out with my nRF24, but wasn't able to replicate the issue. Unless all the information from the template can be provided, there's nothing I can do.

closed 2021-09-27 by jgromes