jgromes/RadioLib #372: [SX1278] Can't transmit LoRa signal with bandwidth less than 62.5 kHz

notsleep196, 2021-09-22:
Hello! I have a problem with transmitting data via SX1278 and bandwidth less than 62.5 kHz.

Im trying to send message using esp8266 and sx1278 and SX127x_Transmit.ino example with LoRa parameters: 

> frequency: 436.7 MHz, 

> bandwidth: 41.7 kHz, 

> sf: 10, 

> cr: 5

I can't receive anything using another esp8266 + SX1278 assembly with same LoRa parameters. I addition, i can't see any transmission using LimeSDR mini and GNU Radio, so i think the module doesn't send anything, but transmit() method return success.

Example initialization of transmitter/receiver:
`int state = radio.begin(436.7, 41.7, 10, 5);`

**Additional info:**
 - MCU: ESP8266
 - Wireless module type SX1278
 - Library version 4.5.0

jgromes, 2021-09-22:
Well spotted, thanks!

This was the result of two things going wrong at once in an attempt to fix issues from the SX1278 errata note. First, to mititgate spurious receiver responses, the errata calls for an offset to teh carrier frequency by the bandwidth value (so if the bandwidth is 41.7 kHz, the RF frequency should be increased by 41.7 kHz). In the code, it was actually being received by 41.7 MHz.

The second issue was that the errata fix was being applied regardless of whether the current operation is transmission or reception - it should bprobably be only done for receiving.

So in the end, it turns out your module does actually transmit - just not at 436.7 MHz where you expected it, but instead at 478.4 MHz. Regardless, it's fixed now.

Thanks for reporting!

closed 2021-09-22 by jgromes

notsleep196, 2021-09-23:
Thank you for fast reaction!
Yes, now i see transmition at 436.7 MHz, but i still can't receive anything by another esp8266 and SX1278 with same initial parameters. I think it's rather strange that the frequency for reception is shifted from transmitter frequency. 
To receive i use SX127x_Receive_Interrupt.ino and it can't receive anything with bandwidth less than 62.5...

jgromes, 2021-09-23:
>  I think it's rather strange that the frequency for reception is shifted from transmitter frequency.

The errata fix is only supposed to shift the intermediate frequency.

The fact you can't receive anything can be caused by your oscillator - are you using a crystal or a TCXO? SX1278 datasheet recommends using TCXO for low bandwidths, as the frequency error of simple crystals (like those used on nearly all cheap SX1278 modules) can be large enough for the low-bandwidth signal to no longer meet the carrier frequency tolerance.

You could try to manually tune your transmitter so that the transmission can be received even with the uncompensated crystal, but keep in mind that the crystal frequency will drift with temperature.

notsleep196, 2021-09-23:
Thank you for response!
I'm using Ra-01 module, so it doesn't have TCXO. I tried to tune frequency to minimize frequency error measured by module. When frequency error is about 50-150 Hz i can receive data!
So, if with small BW parameters link between modules is so strongly dependent on frequency shift between transmitting and receiving modules, what about doppler compensation of about +-20% of BW frequency (as the documentation says)? How will it work?

jgromes, 2021-09-24:
50 - 150 Hz seems way too narrow, what's the behavior at higher bandwidths?

notsleep196, 2021-09-30:
Hello!
I did some tests and i can say that it was my mistake, everything works well now with adeqate frequency shifts.
Thank you for help!