jgromes/RadioLib #374: Reset after Initializing SI4432 and nodmcu

toosimit, 2021-09-28:
First of all thank you for this versatile library. I have problem with esp8266 and SI4432, after running every example, I get same result . the chip not found "error -2" and than esp8266(NODEMCU) reset .
I use Arduino 1.8.13
My wiring is 
SDO > GPIO13
SDI > GPIO12
SCK > GPIO14
nSEL > GPIO15
nIRQ > GPIO5
SDN > GPIO4

I've only change this section of si443x_setting 

// Si4432 has the following connections:
// nSEL pin:  15
// nIRQ pin:  5
// SDN pin:   4
Si4432 radio1 = new Module(15, 5, 4);

and the error that I recived is 
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
3⸮⸮9⸮z	⸮[Si4432] Initializing ... failed, code -2

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffdd0 end: 3fffffc0 offset: 01a0
3fffff70:  fffffffe 3ffee2f8 3ffee458 402010ed  
3fffff80:  00000010 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  3fffdad0 00000000 3ffee480 40204208  
3fffffb0:  feefeffe feefeffe 3ffe84e0 40100ff9  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

Any way to fix this? Did I missed something or is something not right in the Library?
Thanks already :)

jgromes, 2021-09-29:
The reason for the reset is clear - it's the watchdog being triggered in the infinite loop that's caused by initialization failing, it even says so:

```
Soft WDT reset
```

Error code -2 is ERR_CHIP_NOT_FOUND, which means that RadioLib was unable to find the Si4432 module. That's either bad pinout, bad module, or bad wiring - most likely the last option is the root cause. Debug mode output might give us bit more information, but I suggest double-checking your wiring.

toosimit, 2021-09-29:
thank you for answer 
I checked wiring and read template but I'll do it again.
Is there any configuration to define spi pin? I use default spi pin out (esp8266)  maybe I should change to another

toosimit, 2021-09-29:
I found the problem
the data sheet that I have was chines and the miso mosi  was wrong

closed 2021-09-29 by jgromes