jgromes/RadioLib #375: LoRa RFM95 Not Working with Arduino Nano 33 BLE Sense

aperspace, 2021-09-30:
**IMPORTANT: Check the wiki**
Before submitting new issue, please check the [Wiki](https://github.com/jgromes/RadioLib/wiki) and the [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there. 

**Describe the bug**
A clear and concise description of what the bug is. When applicable, please include [debug mode output](https://github.com/jgromes/RadioLib/wiki/Debug-mode).

I am using the Arduino Nano 33 BLE Sense with a Adafruit RFM95 LoRa 868 or 915 MHz. The nano does not work with the LoRa board. 

I have tried initializing the board by using the default SX1278 module and the RFM95 module but they do not work. Below is how I used them: 
RFM95 radio = new Module(4, 3, 2, RADIOLIB_NC);
SX1278 radio = new Module(4, 3, 2, RADIOLIB_NC);

Specifically, the sketch, RadioLib SX127x Transmit Example, is getting stuck on "int state = radio.begin();". The error message in the debug mode is the following: "SX127x not found! (1 of 10 tries) SX127X_REG_VERSION == 0x0000, expected 0x0011" and continues onto the 10th try. 

**To Reproduce**
Minimal Arduino sketch to reproduce the behavior. Please user Markdown to style the code to make it readable (see [Markdown Cheatsheet](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)).

```C++
// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS (Chip select) pin:   10
// DIO0 pin (GO/Interrupt):  2
// RESET (RST) pin: 9
// DIO1 pin:  3
//SX1278 radio = new Module(10, 2, 9, 3);

//RFM95 radio = new Module(Chip_Select, IRQ/GO; Interrupt Pin, RST/Rest Pin)
RFM95 radio = new Module(4, 3, 2, RADIOLIB_NC); // GETS STUCK ON INITIALIZING (SX127x not found! (1 of 10 tries) SX127X_REG_VERSION == 0x0000, expected 0x0011)

//SX1278 radio = new Module(4, 3, 2, RADIOLIB_NC); // GIVES ERROR CODE -2 (https://jgromes.github.io/RadioLib/group__status__codes.html); Radio chip not found during initialization (SX127x not found! (1 of 10 tries) SX127X_REG_VERSION == 0x0000, expected 0x0012)
  
// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);
  while(!Serial) {;}

  Serial.println("GO!");

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... ")); // GETTING STUCK HERE AT THE MOMENT, OBJECT HAS NOT BE INITALIZED CORRECTLY, WHY?
  int state = radio.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */
}

void loop() {
  Serial.print(F("[SX1278] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX127x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}
```
**Expected behavior** 
A clear and concise description of what you expected to happen.

The LoRa RFM95 should transmit data packets. 

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional info (please complete):**
 - MCU: [e.g. Arduino Uno, ESP8266 etc.] Arduino Nano 33 BLE Sense 
 - Link to Arduino core: https://github.com/arduino/ArduinoCore-mbed
 - Wireless module type: SX1278
 - Arduino IDE version: 1.8.16
 - Library version: 4.5.0

jgromes, 2021-09-30:
> SX127x not found! (1 of 10 tries) SX127X_REG_VERSION == 0x0000, expected 0x0011

That suggest incorrect wiring of the SPI bus, you should check that.

aperspace, 2021-10-01:
You were correct, by accident, I had the CS in the code defined as pin 4 (D4) and wired it into pin 10 (D10). However, after I fixed it, it is still throwing me the same error. I checked the rest of my pins. Does the library only accept a certain CS pin from the Nano 33 BLE sense for RFM95s?

jgromes, 2021-10-01:
> Does the library only accept a certain CS pin from the Nano 33 BLE sense for RFM95s?

No, it can be any pin that can function as output.

Checking Nano 33 BLE documentation, it says it's a 3.3V logic device. However, you seem to be powering the LoRa module (including its builtin level shifter) with 5V. So you're sending 5V signals from the LoRa module to the 3.3V Arduino - does it have 5V-tolerant pins?

aperspace, 2021-10-06:
I switched the VIN pin of the LoRa RFM95 from 5V to 3.3V and same error. I did go thru Adafruit's LoRa RFM95 article and it does say explicitly that 5V pin would work fine. 

https://learn.adafruit.com/adafruit-rfm69hcw-and-rfm96-rfm95-rfm98-lora-packet-padio-breakouts/arduino-wiring

The data sheet (https://cdn-learn.adafruit.com/downloads/pdf/adafruit-rfm69hcw-and-rfm96-rfm95-rfm98-lora-packet-padio-breakouts.pdf) also says "Each radio comes with some header, a 3.3V voltage regulator and levelshifter that can handle 3-5V DC power and logic so you can use it with 3V or 5V devices."

aperspace, 2021-10-06:
I got it to work! I will post my code below. 

I simply changed the definition of the module from RFM95 to SX1278 by using the following line: 
SX1278 radio = new Module(10, 3, 2);

And after all the other changes, it works now. 

```C++
// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS (Chip select) pin:   10
// DIO0 pin (GO/Interrupt):  2
// RESET (RST) pin: 9
// DIO1 pin:  3
//SX1278 radio = new Module(10, 2, 9, 3);

//RFM95 radio = new Module(Chip_Select, IRQ/GO; Interrupt Pin, RST/Rest Pin)
//RFM95 radio = new Module(10, 3, 2, RADIOLIB_NC); // GETS STUCK ON INITIALIZING (SX127x not found! (1 of 10 tries) SX127X_REG_VERSION == 0x0000, expected 0x0011) 
//RFM95 radio = new Module(10, 3, 2);
//SX1278 radio = new Module(10, 3, 2, RADIOLIB_NC); // GIVES ERROR CODE -2 (https://jgromes.github.io/RadioLib/group__status__codes.html); Radio chip not found during initialization (SX127x not found! (1 of 10 tries) SX127X_REG_VERSION == 0x0000, expected 0x0012)
SX1278 radio = new Module(10, 3, 2);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);
  while(!Serial) {;}

  Serial.println("GO!");

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... ")); // GETTING STUCK HERE AT THE MOMENT, OBJECT HAS NOT BE INITALIZED CORRECTLY, WHY?
  int state = radio.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */
}

void loop() {
  Serial.print(F("[SX1278] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX127x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}
```

jgromes, 2021-10-06:
> it does say explicitly that 5V pin would work fine

I think you misunderstood my previous comment - the LoRa module will work with a 5V device, however, your Arduino is not a 5V device.

> I simply changed the definition of the module from RFM95 to SX1278

RFM95 is nearly identical to SX1278, but they have a very different frequency range, so make sure your code is matching your hardware. The LoRa chip should be marked.

> SX1278 radio = new Module(10, 3, 2);

That's identical to `SX1278 radio = new Module(10, 3, 2, RADIOLIB_NC);`, so it seems unlikely it should have an effect.

In any case, there doesn't seem to be an issue in the library, so I'm closing this for now.

closed 2021-10-06 by jgromes

aperspace, 2021-10-30:
Yep that was it! Thanks for all the help, it works completely fine now. Sorry for the very rookie questions but I learned a lot from the library.