jgromes/RadioLib #378: Allowed frequency range for sx1278

mputnieks, 2021-10-02:
I noticed that the documentation when hovering over the begin() method for the example transmit code of SX1278 says that the allowed frequency range is 137MHz to 525MHz, yet I was sure that the SX1278 is made for 915MHz range, so why can I not use the 915MHz transmission frequency?
[image]
```
void setup() {
  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));  
  int state = radio.begin();    // <---
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}
```

jgromes, 2021-10-02:
> I was sure that the SX1278 is made for 915MHz range

The SX1276/77/78/79 datasheet seems to disagree with you:
[image]

Allowed ranges for all parameters are taken directly from official documentation, so I'm fairly confident they should be correct.

closed 2021-10-02 by jgromes

mputnieks, 2021-10-02:
True, it sure does, which is even more confusing as I ordered the HopeRF RFM95W-915S2 LoRa(TM) Transceiver from uptronics:
https://store.uputronics.com/index.php?route=product/product&product_id=97&search=RFM95
here's the datasheet about frequencies:
[image]
And, as I examined my module I found that what I got delivered instead was the RFM96W version! I did not realize that before, even as I had noted that it has the sx1278, as I never doubted that I had the right thing.
Anyhow, thank you very much for your time!

Best,
Miķelis