jgromes/RadioLib #397: Cubecell HTCC-AB02 compiling error

sheggy012, 2021-10-30:
**IMPORTANT: Before submitting an issue, please check the following:**
1. **Read [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. RadioLib has a [Wiki](https://github.com/jgromes/RadioLib/wiki) and an extensive [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there.
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/RadioLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about wireless modules, wiring, usage, etc. Only create issues for problems specific to RadioLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

**Sketch that is causing the module fail**

```c++
/*
   RadioLib SX126x Transmit Example
   This example transmits packets using SX1262 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)
   Other modules from SX126x family can also be used.
   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem
   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/
#include "Arduino.h"
#define RADIOLIB_SOFTWARE_SERIAL_UNSUPPORTED
#define RADIOLIB_HARDWARE_SERIAL_PORT Serial1
#define __ASR6501__
#include <RadioLib.h>

// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
SX1262 radio = new Module(10, 2, 3, 9);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1262 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize SX1262 with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */
}

void loop() {
  Serial.print(F("[SX1262] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
[... 100 lines cut ...]
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Sergej\\AppData\\Local\\Temp\\arduino_build_447486\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Sketch wird kompiliert...
"C:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\tools\\gcc-arm-none-eabi\\8-2019-q3/bin/arm-none-eabi-g++" -DREGION_EU868 -DCubeCell_BoardPlus -DARDUINO=10812 -DACTIVE_REGION=LORAMAC_REGION_EU868 -DLORAWAN_DEVEUI_AUTO=0 -DLORAWAN_CLASS=CLASS_A -DLORAWAN_NETMODE=true -DLORAWAN_ADR=true -DLORAWAN_UPLINKMODE=true -DLORAWAN_NET_RESERVE=false -DAT_SUPPORT=1 -DLoraWan_RGB=1 -DLoRaWAN_DEBUG_LEVEL=0 -DSOFT_SE -DCY_CORE_ID=0 -DCONFIG_LORA_USE_TCXO "-DCONFIG_MANUFACTURER=\"ASR\"" "-DCONFIG_DEVICE_MODEL=\"6501\"" "-DCONFIG_VERSION=\"v4.0\"" -DARDUINO_ARCH_ASR650X -D__asr650x__ -D__ASR6502__ -DF_CPU=48000000L "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/board/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/board/src/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/board/inc/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/lora/radio/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/device/sx126x/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/loramac" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/loramac/region/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/lora/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/lora/system/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/lora/system/crypto/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/port/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/port/include/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/projects/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/projects/PSoC4/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/cores/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/Serial/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/Wire/" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0/cores/asr650x/SPI/" -mcpu=cortex-m0plus -mthumb -w -Wall -g -gdwarf-2 -MMD -Os -mapcs-frame -mthumb-interwork -fno-common -ffat-lto-objects -ffunction-sections -fdata-sections -fno-builtin-printf -fno-builtin-sprintf -fno-builtin-snprintf -fno-builtin-fflush -Wno-strict-aliasing -c -fno-exceptions -fexceptions -fno-rtti "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0\\cores\\asr650x" "-IC:\\Users\\Sergej\\AppData\\Local\\Arduino15\\packages\\CubeCell\\hardware\\CubeCell\\1.3.0\\variants\\CubeCell-BoardPlus" "-IC:\\Users\\Sergej\\Documents\\Arduino\\libraries\\RadioLib\\src" "C:\\Users\\Sergej\\AppData\\Local\\Temp\\arduino_build_447486\\sketch\\sketch_oct29a.ino.cpp" -o "C:\\Users\\Sergej\\AppData\\Local\\Temp\\arduino_build_447486\\sketch\\sketch_oct29a.ino.cpp.o"
In file included from C:\Users\Sergej\Documents\Arduino\libraries\RadioLib\src/RadioLib.h:45:0,

                 from C:\Users\Sergej\Documents\Arduino\sketch_oct29a\sketch_oct29a.ino:18:

C:\Users\Sergej\Documents\Arduino\libraries\RadioLib\src/Module.h:427:12: error: declaration does not declare anything [-fpermissive]

     static void yield();

            ^

Bibliothek RadioLib in Version 4.6.0 im Ordner: C:\Users\Sergej\Documents\Arduino\libraries\RadioLib  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board CubeCell-Board Plus（HTCC-AB02）.
**Additional info (please complete):**
 - MCU: Heltec Cubecell  HTCC AB-02
 - Link to Arduino core: https://github.com/HelTecAutomation/CubeCell-Arduino
 - Wireless module type ASR6502 / SX1262 
 - Arduino IDE version  1.8.12
 - Library version 4.6.0

Hello all, 
2 days ago the Cubecell module was added to the supported modules. However, I just can't get it to compile. I use a simple Example Sketch. I have only defined 4 single lines in the top. Don't know if it's right. I hope someone can help. Would be happy to be able to use the library on the module

jgromes, 2021-10-30:
I was unable to replicate this, couple of thoughts:

1. Please don't add the extra definitions, you will confuse the platform detection.
2. Which version of the CubeCell core are you running? Myself (and the automated compilation in the repository, which is done prior to every release for every single platform) is using the latest CubeCell version available from Arduino IDE boards manager (1.2.0).

sheggy012, 2021-10-30:
Okay thanks for the info. I have used the version 1.3.0. Now changed to 1.2.0 and look it runs! Thanks a lot!

With version 1.3.0 without the difines I get the message: ``C:\Users\Sergej\Documents\Arduino\libraries\RadioLib\src/Module.h:8:28: fatal error: SoftwareSerial.h: No such file or directory ``
That's why I added those above.

jgromes, 2021-10-30:
If support for SoftwareSerial is about to be dropped from CubeCell, then it's probably better to just remove it from RadioLib now and save some trouble in the future.

I'll also take a look at that error you had for further future-proofing.

jgromes, 2021-10-30:
Installed version 1.3.0 and managed to replicate the issue you had, which is due to the fact that with every version, CubeCell is even more broken than before. Honestly it's a stretch to call that pile of junk "Arduino-compatible" ...

Anyway, future issues resolved, 1.3.0 should work now too.

closed 2021-10-30 by jgromes