jgromes/RadioLib #400: [Announcement] RadioLib 5.0.0 changes

jgromes, 2021-11-07:
# This is a list of upcoming changes in RadioLib 5.0.0

Major changes will be done in RadioLib 5.0.0 release, so I'm posting these publicly to maybe get some feedback.

## Breaking changes

- [ ] Add prefix `RADIOLIB_` to all macros defined by the library (including status codes).

This is already the case for most cases, but it should be done everywhere, especially status codes.

## Non-breaking changes

- [ ] Drop support for ESP8266 module (not platform)

Rationale: ESP8266 is probably the least used module in here. It's also causing issues when compiling on some platforms (ESP8266 and anything using mbed), which is currently handled by exclusion macros. It also needs AT command set support, and doesn't really fit with the general ham radio use-case. 

- [ ] Drop support for HTTP and MQTT protocols (as well as TransportLayer)

Rationale: Similar to the point above. There's also libraries that do a better job supporting these protocols.

- [ ] Drop support for HC05 and JDY08 (as well as ISerial)

Rationale: Also very rarely used; the RadioLib layer is also just a thin wrapper around Serial interface, which doesn't add much new. Just two relicts of the past.

- [ ] Drop supprot for XBee

Rationale: Again, rarely used module, and other libraries probably do a better job.

- [ ] Introduce Arduino-independent hardware abstraction layer

Rationale: It's currently not possible to use RadioLib without Arduino. Almost all references to Arduino API are contained in Module.cpp, but running RadioLib on non-Arduino platform would still require re-implementing things like `SPIClass`.  The `Module` class should be reworked to use a set of callbacks for Arduino API calls (like SPI write or digital read). If compiling for Arduino, the callbacks would be pre-configured to use calls from Arduino API. If compiling for something other than Arduino, the user would have to set all the callbacks. This preserves easy use on Arduino, but makes it portable to nearly anything with all the porting done in user code.

jgromes, 2021-11-07:
cc @phretor @4m1g0 @geeksville

You guys all have major projects using RadioLib, so I'm letting you know in advance. Let me know if you have any feedback or other suggestions.

geeksville, 2021-11-08:
@jgromes that seems like a good list and we'll happily update to it after it is out.  Thank you for your efforts!

Two comments:

* I've been meaning to send in a PR with a small number of changes for the sx12XX drivers to allow use of those drivers across  processor deep sleep.  These changes are only needed for very low power applications the main CPU can be completely shutdown and then awoken by a RX packet (booting from scratch) and then non-destructively reiniting the radio chips well enough to resume normal operations (and process the rx packet).  I'll try to do this in the next month or so.

* re: adding an abstraction layer in order to be less 'Arduino' dependent.   I think this is a great idea.  And what you describe is probably better than the approach I needed to take for meshtastic - where I wanted to run with numerous other arduino libs on either arduino (nrf52 or esp32 really) systems or posix systems (linux, os-x, windows).  In case it is useful for you, my young attempt at that abstraction layer is here: [portduino](https://github.com/geeksville/framework-portduino )
This library is more about 'running any arduino code on top of any Posix OS' though...

jgromes, 2021-11-14:
The new API for version 5.0.0 is now public on https://github.com/jgromes/RadioLib/tree/hal

@geeksville thanks for pointing me to portduino! In the end, I stuck to the original idea of callbacks, which unfortunately took a lot more preprocessor wizardry thank I would have liked. Anyway, RadioLib should be able to run on virtually anything that has a C++ compiler now.

jgromes, 2021-11-27:
5.0.0 is now released (https://github.com/jgromes/RadioLib/releases/tag/5.0.0).

closed 2021-11-27 by jgromes