jgromes/RadioLib #408: First byte in the byte array received is the length of the array

jbirmingham40, 2021-11-17:
I tested out your RF69_Receive and RF69_Transmit examples using the commented out code for sending byte arrays.  The byte array returned does not match the one sent.  After some debugging it appears that you are prepending the size before sending the array and not removing it before returning the array.  If I skip the first element of the array I get what I expect however this is probably not what you intended.

jbirmingham40, 2021-11-20:
Something else is going on.  It seems that I have to send in an array of 64 bytes in order to get the correct results.  I worked around the issue.  You can close this.

closed 2021-11-20 by jbirmingham40

jgromes, 2021-11-21:
> prepending the size before sending the array

That is intentional, and is how RF69 operates in variable packet length mode (which is the default).

However, there was indeed a bug in the logic resolving the number of actually received bytes vs. the number of bytes user wants to read. It should be fixed now, thanks for reporting.