jgromes/RadioLib #409: Question: sx126x channel activity detection

sh123, 2021-11-20:
Before transmitting would like to check if channel is busy on sx126x, what would be the correct approach to that on sx126x module?

closed 2021-11-20 by sh123

jgromes, 2021-11-20:
I'm guessing that you found the two channel activity detection examples for SX126x.

sh123, 2021-11-20:
No, it is exactly what is needed, it is more for monitoring, not really suitable for CSMA, need to quickly check for any activity on the channel before deciding if need to transmit or not without initiating scan.

jgromes, 2021-11-20:
Unforuntately, for LoRa transmissions, you have to use CAD. The reason as to why you can't use something simpler like RSSI measurement is that LoRa signals can be below the noise floor. Below SNR of 0 dB, you will not see difference in RSSI between LoRa transmission and noise.

In theory, you could use the interrupt-driven CAD, and if you don't see a transmission within some arbitrary timeout (e.g. 1 second), you consider the channel free.

sh123, 2021-11-20:
It is understandable, maybe example could include reading and writing actual data, where it will transmit when channel is free and will read data if packet is detected, from example it is not obvious if such scenario is allowed.