jgromes/RadioLib #411: Problem AFSK SX1278

jp-polito, 2021-11-25:
Hi! I'm trying to run the AFSK example in my SX1278 module it works but I can't decode the signal. It seems the module is active all the time. I can't understand what's going on.
[image]

jgromes, 2021-11-25:
> I can't understand what's going on

Neither can I if you don't tell me what are you using to decode. Soundmodem should work, but I think that's Windows-only.

jp-polito, 2021-11-25:
II'm trying to use GQRX's own decoder.

jp-polito, 2021-11-25:
Is it normal for the module to be activated all the time? I'm sending some prints.
[image]
[image]

jgromes, 2021-11-26:
> Is it normal for the module to be activated all the time

No, that shouldn't happen - I pushed a commit which should fix that.

> I'm trying to use GQRX's own decoder.

OK, do you know how it works? Because I was never able to get it to work with known good AX.25 frames.

jp-polito, 2021-11-26:
I've already run some tests using gqrx and AFSK on other circuits and it worked. And about the carrier I used a logic level converter from 5V to 3.3V

jgromes, 2021-11-27:
I tried this in a VM running Ubuntu 18.04 and wasn't able to replicate the issue using Arduino Uno and SX1278.

[image]

Pretty much default settings, just changed mode to NFM and adjusted filter width.

closed 2021-11-27 by jgromes