jgromes/RadioLib #412: Pinouts for Heltec Cubecell ASR650x

plainolddave, 2021-11-27:
Hi Jan - great work on RadioLib - currently I'm migrating from another library (*cough* RadioHead) and had a painful time integrating RadioLib with the Heltec Cubecell ASR-650x series (either ASR-6501 or ASR-6502)

Luckily it was partly solved by #397 (thank you) and I'm raising this new issue to make anyone else aware there is an easy but non-obvious way to understand the correct pin configuration for RadioLib & Cubecell.

Suggest including as a note in the examples for SX1262, that when defining the radio object:
- include the Cubecell "board-config.h" from ...\Arduino15\packages\CubeCell\hardware\CubeCell\1.3.0\cores\asr650x\board\inc\board-config.h
- then use the pre-defined pinouts to initialise RadioLib:

```
#include <RadioLib.h>
#include "board-config.h"
SX1262 radio = new Module(RADIO_NSS, RADIO_DIO_1, RADIO_RESET, RADIO_BUSY);
```

Seems to work fine with current/latest versions at the time of raising this issue:
- MCU: [Heltec Cubecell GPS-6502 (htcc-ab02s)](https://heltec.org/project/htcc-ab02s/)
- Wireless module type ASR6502 / SX1262
- Link to Arduino core: https://github.com/HelTecAutomation/CubeCell-Arduino
- Arduino IDE version 1.8.16 / Visual Micro 21.11.08.0
- RadioLib version 4.6.0 (including #397)

Hope this helps

jgromes, 2021-11-27:
Thanks for pointing this out. Yes, you have to know the pins that are being used internally by CubeCell if you want to use the built-in SX1262. That's simply because RadioLib was written to be as generic as possible.

Considering the number of hoops I had to jump through in order to get CubeCell working at all (*cough* pinMode as macro)(*cough* yield as macro that does bugger all)(*cough* outdated GCC), stramlining this a bit more can't hurt. So I added the board_config include and pin definition to BuildOpt and now you can skip it all by calling:

```c++
SX1262 radio = new Module(RADIOLIB_BUILTIN_MODULE);
```

CubeCell may not be the only platform with a built-in module, so it might be better to have a generic option to specific fixed pins in the library itself, so that users don't have to figure it out.

closed 2021-11-27 by jgromes

plainolddave, 2021-11-28:
Great - thanks Jan - appreciated. I was reading comments in radiolib source code and laughed :) completely agree with you. On the plus side though their hardware has a lot in a small package (and is reasonably priced) so on balance it works for now. Thanks again