jgromes/RadioLib #413: Buffer overrun on ESP32

icanhazpython, 2021-11-29:
**Describe the bug**
Disclaimer - I'm not extremely well-versed in c/c++ or arduino's version of it. When using an SX1276 on an ESP32 and trying to receive data using arduino "strings", I get what appears to be a buffer overflow. Only when I predefined the size of the string do I not have an issue. I looked a bit at the source, and (disclaimer applies here) it seems like the code attempts to receive an arbitrary length of data and write it to an array with a size of 1 (when using 0 for the receive function's 2nd parameter): https://github.com/jgromes/RadioLib/blob/df466486aab98b7a982c5ae73e75194576c0dc0e/src/protocols/PhysicalLayer/PhysicalLayer.cpp#L100

Here's the error I see over the serial monitor:
```
CORRUPT HEAP: Bad tail at 0x3ffbdb4d. Expected 0xbaad5678 got 0x30313044
abort() was called at PC 0x40083b99 on core 1
ELF file SHA256: 0000000000000000
Backtrace: 0x400857b4:0x3ffb1ab0 0x40085a29:0x3ffb1ad0 0x40083b99:0x3ffb1af0 0x40083cc5:0x3ffb1b20 0x400dc0cb:0x3ffb1b40 0x400d853d:0x3ffb1e00 0x400d84d8:0x3ffb1e50 0x4008a0b5:0x3ffb1e80 0x40081dee:0x3ffb1ea0 0x40083a91:0x3ffb1ec0 0x4000bec7:0x3ffb1ee0 0x400ea9bd:0x3ffb1f00 0x400ea94d:0x3ffb1f20 0x400d31db:0x3ffb1f40 0x400d0ffe:0x3ffb1f70 0x400d62b5:0x3ffb1fb0 0x40086a39:0x3ffb1fd0
```

**To Reproduce**
```
String str;
int state = radio.receive(str, 0);
```

**Expected behavior**
Receive the data of an unknown size into a String.

**Additional info**
 - MCU: ESP32
 - Wireless module type: SX1276
 - Arduino IDE version : 1.8.16
 - Library version: 5.0.0

icanhazpython, 2021-11-29:
I was able to work around this by using this code. Probably inefficient, and obviously has a max buffer size, but it works for my application. Would be nice to know how to make the automatic string allocation work. 
```
String str = String(80);
int packetLen;
char *strNull;
int state = radio.receive(str, 80);

if (state == RADIOLIB_ERR_NONE)
{
    packetLen = radio.getPacketLength();
    strNull = new char[packetLen];
    str.toCharArray(strNull, packetLen);
    Serial.println(strNull);
}
```

jgromes, 2021-11-29:
You're right, looks like this got broken in a recent commit. It should be fixed now.

closed 2021-11-29 by jgromes