jgromes/RadioLib #416: CUSTOM SPI PINS on ESP32

PacciK, 2021-12-02:
Hi @jgromes , I have updated to 5.0.0 and in my testing, I am using two LORA modules on the two hardware SPI's available on the ESP32: SPI_V and SPI_H.
earlier reference to the method for using custom pins : https://github.com/jgromes/RadioLib/issues/252

On the first compile with 5.0,0 this error is being presented by Arduino 1.8.16:
`error: no matching function for call to 'Module::Module(int, int, int, int, SPIClass&)`

Further down the error messages in the IDE,:
`Arduino\libraries\RadioLib\src/Module.h:49:5: note:   candidate expects 6 arguments, 5 provided`

which is referring to:

`  
    Module(RADIOLIB_PIN_TYPE cs, RADIOLIB_PIN_TYPE irq, RADIOLIB_PIN_TYPE rst, RADIOLIB_PIN_TYPE gpio, SPIClass& spi, SPISettings spiSettings);
`

Can you please suggest a suitable example constructor to use ?

jgromes, 2021-12-02:
There were some API changes in 5.0.0, one of them removed a lot of the constructors, resolving some ambiguities. However, it seems that one of the things that got removed was the default SPISettings parameter, which is incorrect. Should be fixed in the latest commit.

closed 2021-12-02 by jgromes

PacciK, 2021-12-02:
Thanks @jgromes . Will check it out

PacciK, 2021-12-02:
Yes adding what you amended in https://github.com/jgromes/RadioLib/commit/97f3b94c4f5d13a1add3c6609e89768b32585a20
has fixed the compilation issue.

Last error to fix:
What can I use instead of 
`if (radio.setTCXO(1.8) == ERR_INVALID_TCXO_VOLTAGE){`

the error coming up in the IDE is
exit status 1
'ERR_INVALID_TCXO_VOLTAGE' was not declared in this scope

jgromes, 2021-12-02:
As a part of the API changes, all macros defined by the library now start with RADIOLIB_. So use RADIOLIB_ERR_INVALID_TCXO_VOLTAGE.

PacciK, 2021-12-02:
Amazing @jgromes 
I can now continue playing.
Thanks a million !