jgromes/RadioLib #418: FSK connection not working with SX1276

pete20211207, 2021-12-07:
Setup:
* Arduino Uno
* with a selfmade Arduino Radio Shield (according Rev. C)
* **SX1276** module
* RadioLib Rev. 5.0.0 / Rev.4.6.0
* **LoRa connection is working well** with different configurations

I tried to get a **FSK connection** but the receiver side always runs into **timeout**.
TX and RX setup() e.g.

```
int state = radio.beginFSK(); // RX fails
// or
int state = radio.beginFSK(869.0, 4.8, 5.0, 125.0, 10, 16, false); // RX fails
// or
int state = radio.beginFSK(869.0, 300.0, 50.0, 200.0, 6, 16, false); // RX fails
uint8_t syncWord[] = {0xA5};
state = radio.setSyncWord(syncWord, 1);

```

TX loop():
`
int state = radio.transmit(txString); // TX working well
`

RX loop():
`
int state = radio.receive(str); // RX runs into timeout
`

I tried also other syncWord[] settings : same result

Could you provide some hints on a working FSK connection?
Thank you a lot.

jgromes, 2021-12-07:
FSK messages are a lot shorter than LoRa - try to use interrupt receive.

pete20211207, 2021-12-09:
Same results with interrupt receive: LoRa is working well - FSK does not reicevie anything.

jgromes, 2021-12-12:
I was able to replicate the issue, turns out this was caused by the fix for #402.

Should be fixed now, thanks for reporting.

closed 2021-12-12 by jgromes