jgromes/RadioLib #419: LoRa connection not working with SF6 on SX1276

pete20211207, 2021-12-07:
Hence LoRa connection with SF7 to SF12 are working well for my setup with SX1276, I also tried to use SF6.
But communication with SF6 leads into "empty data strings" on the receiver side, i.e. I so receive something without CRC error, but an empty string.

I learned that SF6 needs "implicit header mode".
How to that done for SX1276?

jgromes, 2021-12-07:
> I learned that SF6 needs "implicit header mode".

Very nice! Are you aware of the other limitations? Such as the fact that you need to know the length of data to be received in advance?

> How to that done for SX1276?

I suspect the method `SX1278::implicitHeader(size_t len)` might be useful here.

pete20211207, 2021-12-10:
Well, all the settings mentioned in the SX1276/78 datasheet are done in SX1278.cpp by 
```
int16_t SX1278::setSpreadingFactorRaw(uint8_t newSpreadingFactor) {
  // set mode to standby
  int16_t state = SX127x::standby();

  // write registers
  if(newSpreadingFactor == RADIOLIB_SX127X_SF_6) {
    state |= _mod->SPIsetRegValue(RADIOLIB_SX127X_REG_MODEM_CONFIG_1, RADIOLIB_SX1278_HEADER_IMPL_MODE, 0, 0);
    state |= _mod->SPIsetRegValue(RADIOLIB_SX127X_REG_MODEM_CONFIG_2, RADIOLIB_SX127X_SF_6 | RADIOLIB_SX127X_TX_MODE_SINGLE | (SX127x::_crcEnabled ? RADIOLIB_SX1278_RX_CRC_MODE_ON : RADIOLIB_SX1278_RX_CRC_MODE_OFF), 7, 2);
    state |= _mod->SPIsetRegValue(RADIOLIB_SX127X_REG_DETECT_OPTIMIZE, RADIOLIB_SX127X_DETECT_OPTIMIZE_SF_6, 2, 0);
    state |= _mod->SPIsetRegValue(RADIOLIB_SX127X_REG_DETECTION_THRESHOLD, RADIOLIB_SX127X_DETECTION_THRESHOLD_SF_6);
  } ...
```
but always received an empty string.
Any hints on the transmission method? e.g. to send "abc" with SF6 and the default 

`int state = radio.begin();`

jgromes, 2021-12-10:
> Well, all the settings mentioned in the SX1276/78 datasheet are done in SX1278.cpp

Not all of them. One cannot be done by the library, as it depends on the user. Recall that "you need to know the length of data to be received in advance".  That translates to having to call `radio.receive()` with the expected number of bytes provided, e.g. `radio.receive(&str, len);`.

> Any hints on the transmission method?

Transmission for SF6 is the same as for any other spreading factor, it's the reception that's changed a bit.

jgromes, 2021-12-12:
I was unable to replicate this, following all proper configuration. On both sides, you have to `setSpreadingFactor(6);` and enable implicit header with `implicitHeader(len);` where `len` is the length of data to be transmitted (12 for the default "Hello World!" example).

In addition, on the receiver side, you have to call `receive(&str, len);`/`startReceive(len);`.

Of course, if you either don't enable the implicit header or don't provide the number of expected bytes to the receive method, you will receive nothing - that's the limitation of SF6.

closed 2021-12-12 by jgromes