jgromes/RadioLib #422: Specifying SPI pins, is it possible to do this?

smartua7, 2021-12-11:
Hello. I have been trying to configure the sx1262 (ebyte E22-900) and esp32 module for several days, but I have a problem to configure the SPI pins. My questions are: Will the library work on ESP32-C3 (RiscV 1-Core)? How to change the settings of pins of a regular ESP32 (wroom, nodemcu). Since it is possible to use 2 SPI (VSPI, HSPI).
I am using given examples in Arduino IDE:
_RadioLib\examples\SX126x\SX126x_Transmit
RadioLib\examples\SX126x\SX126x_Receive_

I've tried specifying like this:
```
SPIClass *loraSPI = new SPIClass(HSPI);
SX1262 radio = new Module(5, 4, 16, 17, *loraSPI);
```
but it doesn't work as there is no SPIClass class.
Please tell me how to specify the SPI pins?

jgromes, 2021-12-11:
> Will the library work on ESP32-C3 (RiscV 1-Core)?

Assuming the ESP32 Arduino core supports it, it should work out-of-the-box.

> but it doesn't work as there is no SPIClass class.

SPIClass is a part of the Arduino framework, so it should always be present. Please fill in the [issue template](https://github.com/jgromes/RadioLib/blob/master/.github/ISSUE_TEMPLATE/module-not-working.md). What is the output when you try to compile?

jgromes, 2021-12-14:
#include <SPI.h> is already being done by RadioLib, your Arduino IDE seems broken. Again, please fill in the issue template and post the original error message(s).

> I also have a question about setting the channel number, netid (device id) and key in Lora mode?

I'm assuming you're referring to LoRaWAN - that is not supported by RadioLib.

smartua7, 2021-12-14:
I have already solved all the questions, thank you very much. There are no more mistakes. I am interested in the ability to understand who sent a message between several devices with one sync word and the sending takes place on the broadcast port 65536, maybe I do not quite understand how this works in Lora, but does the library allow me to understand this

jgromes, 2021-12-14:
I don't know what "broadcast port" you're referring to - LoRa packets do not have the concept of port.

closed 2021-12-14 by jgromes

smartua7, 2021-12-14:
> 

I mean radio.set Broadcast Address (0xFF);
Is it possible to find out which device sent the message without indicating this in the load message?

jgromes, 2021-12-14:
That only applies for FSK packets. And no, it's not possible, you would have to include that information in the payload. The address is just a simple filter that the radio applies.

bjfhskj, 2022-06-06:
> Hi, I solved it in the following way. Connected by force: #include <SPI.h> Specified settings
> 
> ```
> SPIClass mySpi (VSPI);
> SPISettings spiSettings (2000000, MSBFIRST, SPI_MODE0);
> SX126x radio = new Module (5, 4, 16, 17, mySpi, spiSettings);
> ```
> 
> And also in the setup () block I forcibly set `mySpi.begin ();` For me, it still remains a mystery how to set specific pins for SPI lines.
> 
> Thanks for answers! I also have a question about setting the channel number, netid (device id) and key in Lora mode? I studied all the commands, but I did not find such commands similar commands not on one of the levels of the library

I follow your code with LLCC68,but get a error:-20;
the debug message:[SX1262] Initializing ... failed, code -20