jgromes/RadioLib #424: Compile error in debug mode

steven52880, 2021-12-18:
- IDE: PlatformIO
- Ver: 5.0

When I discomment `//#define RADIOLIB_DEBUG`, I got a compilation error.

```
In file included from .pio\libdeps\client\RadioLib\src\TypeDef.h:4:0,
                 from .pio\libdeps\client\RadioLib\src\Module.h:4,
                 from .pio\libdeps\client\RadioLib\src\Module.cpp:1:
.pio\libdeps\client\RadioLib\src\Module.cpp: In member function 'void Module::SPItransfer(uint8_t, uint8_t, uint8_t*, uint8_t*, uint8_t)':
.pio\libdeps\client\RadioLib\src\BuildOpt.h:862:78: error: no matching function for call to 'HardwareSerial::print()'
     #define RADIOLIB_DEBUG_PRINT(...) { RADIOLIB_DEBUG_PORT.print(__VA_ARGS__); }
```

jgromes, 2021-12-18:
I waws not able to replicate this in Arduino IDE. What platform are you using? Arduino Uno/ESP32/...

steven52880, 2021-12-19:
I am using nodemcu-32s.
RadioLib@4.6.0 dosn't has the problem.

steven52880, 2021-12-19:
I found that the problem has been fixed in [here](https://github.com/jgromes/RadioLib/commit/1c54c1d1c66a75e182ae0e3dcf91c2b83b535d53#diff-7bba90f01f25deea8207c934b2fbc6f954ab7b726baaaf0c10023c7602977e01).
But only the released version can be downloaded on PlatformIO.

[image]

closed 2021-12-19 by steven52880

jgromes, 2021-12-19:
Thanks for letting me know - in that case, I'm going to release an update, there's been a couple other bugfixes.