jgromes/RadioLib #425: LLCC68 dosn't work in SF11

steven52880, 2021-12-19:
**Additional info (please complete):**
 - MCU: ESP32
 - Wireless module type: Ra01SC(LLCC68)
 - IDE: PlatformIO
 - Library version 5.0.0

**Describe the bug**
Hello. My module can't work while begin with 500k and sf10/sf11.  
`[SX1262] Initializing ... failed, code -20`
I found another [issue](https://github.com/jgromes/RadioLib/issues/315) about it, but the problem dosn't seem to be resolved.
Could you help me check the problem?

**Log**
```
[SX1262] Initializing ... M     SX126x
CMD     80
DATW    0       A2

CMD     80
DATW    0       A2

CMD     8F
DATW    0       A2      0       A2

CMD     8A
DATW    1       A2

CMD     93
DATW    20      A2

CMD     88
DATW    3       A2      18      A2      A       A2      0       A2      0       A2      0       A2      0       A2

CMD     2
DATW    3       A2      FF      A2

CMD     8
DATW    0       A2      0       A2      0       A2      0       A2      0       A2      0       A2      0       A2      0       A2    

CMD     89
DATW    7F      A2

CMD     11
DATR    0       A2      0       1

Symbol length: 4.10 ms
CMD     8B
DATW    B       A2      4       A2      3       A2      0       A2

CMD     11
DATR    0       A8      0       FF

failed, code -20
```

**To Reproduce**
```
radio.begin(434.53, 500.0, 11, 7, '\022', 10, 8, 0);
```

jgromes, 2021-12-20:
You're right, LLCC68 is missing a dedicated `begin`, so it's actually calling `SX1262::begin`, which is the reason why the range checks fail.

I added the missing method, so it should be fixed now.

closed 2021-12-20 by jgromes

steven52880, 2021-12-21:
setBandwidth should be ahead of setSpreadingFactor, or it will exit with -9.
@jgromes

jgromes, 2021-12-22:
You'Re right, that's indeed the case when using non-default parameters for begin(). It should be be fixed now.