jgromes/RadioLib #426: backwards-incompatible changes ?

K4KDR, 2021-12-20:
I'm sure questions from non-programmers are painful - very sorry!

Against all odds I was able to find why an old LoRa RX sketch here would compile w/ v4.6.0 but **not** v5.0.0 and newer:  from your examples in v5.0.0 I found that items such as `ERR_RX_TIMEOUT` had to be changed to `RADIOLIB_ERR_RX_TIMEOUT`.

I also found the following warning on the RELEASE page:

```
Release 5.0.0
Incompatibility warning
This release introduces backwards-incompatible changes to the public API!
```

... when there are updates such as this, is there any kind of **list** as to the items that require changing in code, or should I just look through the examples as I did here?  Or perhaps some trick to identify the changed syntax, etc.?

Many thanks & I wish you a happy Holiday Season!

-Scott,  K4KDR

jgromes, 2021-12-20:
The list of changes in 5.0.0 release is admittedly lacking, the breaking changes should be clearly stated, so that's on me.

However, the changes prior to 5.0.0 were announced in #400, which includes a list of all breaking changes. In this case, only the renaming of constants (`ERR_RX_TIMEOUT` -> `RADIOLIB_ERR_RX_TIMEOUT`) is such a breaking change.

Thanks Scott - you too, Merry Christmas and take care!

closed 2021-12-20 by jgromes

K4KDR, 2021-12-20:
Ahh - I searched 'Issues' but for my error... not for 'Changes' related to v5.0.0 more generally.

Thanks a million for the reply - makes sense now!