jgromes/RadioLib #429: Invalid conversion in new HAL defines in Module.cpp

omegat, 2021-12-23:
Hi I use your lib in a complicated APRS project and was delighted that it now in cludes APRS so I wouldn't have to fix my semi-working Mic-E implementation. It is based on a NRF52832 µC using the stock PlatformIO Arduino stack for simplicity. Long story short. When compiling my project with 5.1.0 i get the following compiler error (Using -DRADIOLIB_TONE_UNSUPPORTED):

`.pio\libdeps\...\RadioLib\src\Module.cpp: In constructor 'Module::Module(uint32_t, uint32_t, uint32_t, uint32_t)':
.pio\libdeps\...\RadioLib\src\Module.cpp:28:42: error: invalid conversion from 'void (*)(uint32_t, voidFuncPtr, uint32_t) {aka 
void (*)(long unsigned int, void (*)(), long unsigned int)}' to 'Module::attachInterrupt_cb_t {aka int (*)(long unsigned int, void (*)(), long unsigned int)}' [-fpermissive]`

The compiler error is fixed by simply changing line 28 and 29 in Module.cpp to:
`  setCb_attachInterrupt((Module::attachInterrupt_cb_t)::attachInterrupt);
  setCb_detachInterrupt((Module::detachInterrupt_cb_t)::detachInterrupt);`
And thus casting the pointers correctly. Don't know if the lib works with this fix yet, but it now compiles fine.

Please understand that I can't supply my project with the problem; it is very delicate to compile (depends on many external libs and so on) and runs on custom hardware.

jgromes, 2021-12-24:
I was not able to replicate this issue. As far as I can tell, everything compiles fine on all supported platforms (since it gets tested by CI prior to any release).

I'm missing important info about your build system. Which nRF52 Arduino are you using? And why are you defining `RADIOLIB_TONE_UNSUPPORTED`?

omegat, 2021-12-24:
> I was not able to replicate this issue. As far as I can tell, everything compiles fine on all supported platforms (since it gets tested by CI prior to any release).

I was afraid that you wouldn't be able to replicate my problem...

> I'm missing important info about your build system. Which nRF52 Arduino are you using? And why are you defining `RADIOLIB_TONE_UNSUPPORTED`?

As I said, it's a homebrew nRF52832 board, using a 'generic NRF52832' board definition (using a variant of a PCA1000X) in PlatformIO, which - I know - makes things next to impossible to reproduce. With the PlatformIO variant of Arduino for NRF52, there is no tone and notone implementation, thus  `RADIOLIB_TONE_UNSUPPORTED` (not defining it throws an error a la "Module.cpp:25:16: error: '::tone' has not been declared". 
> EDIT: On a side note, without `tone` you won't be able to send APRS via AFSK. Also, Mic-E isn't supported yet.

...but sending AFSK with `RADIOLIB_TONE_UNSUPPORTED` defined (SX1276) was possible in a lower version of the lib (don't remember which exactly; sorry). If you like I can send you my not-complete-but-almost-working Mic-E implementation. 
It is also possible to put these chips into FSK-Mode, disable all funny buissenes (like preambles, whitening etc.) and write the 'AFSK' data into the FIFO. With a carefully chosen bitrate you can also emulate AFSK. That way you eliminate the tone-pcoblems and free a GPIO. This was the way the Pecanpico10 balloon sent AFSK / APRS. I tried to implement it that way first but I didn't get it to work. Then I discovered your lib which supported AFSK and started to implement APRS/Mic-E when you released 5.1 with APRS working.

I can try to set up a 'light' version of my build environment to reproduce tomorrow...

Merry Christmas!

jgromes, 2021-12-25:
If you could put together a minimum program to replicate the issue, that would help me a lot.

Regarding AX.25 without AFSK, that is indeed possible by using 2-FSK (and was in fact the only way to do that before I discovered the hack using tone) - there's even examples for it, the ones without the _AFSK suffix. But you have to demodulate in LSB mode, so I think that won't work with APRS infrastructure, which expects NFM, right? I could be wrong here, I'm not all that familiar with HAM stuff.

> If you like I can send you my not-complete-but-almost-working Mic-E implementation

Yes please! APRS support here is very basic at the moment, when I implemented it I just added the most basic stuff. I don't know which parts of APRS are in common use, so decided to extend it only on request.

Merry Christmas to you as well!

omegat, 2021-12-25:
[radiolibtest.zip](https://github.com/jgromes/RadioLib/files/7775904/radiolibtest.zip)
OK this should do it. It fails building of the 'test' environment just as my original project does. As I inspected things a little more closely I think it should be ""easy to fix"":  attach interrupt is a function of the 'type' void (*)(long unsigned int, void (*)(), long unsigned int) - but your macro expects Module::attachInterrupt_cb_t which is equal to **int (*)**(long unsigned int, void (*)(), long unsigned int). So your type expects an int* whereas attachinterrupt returns void*. On the metal it doesn't matter that much if the pointer returned is of the type void* or int* - they're both 32bits in my/this case - but the compiler has to keep track of the types. When you define ´-DRADIOLIB_CUSTOM_ARDUINO´ (which I discovered when sercing for the place where you define these ..._cb_t -types) things get really out of hand.

I can send you my test project where I "developed" the mic-e implementation as a addition to the ax25-class. It contains a lot of wrapping-code to use the implementation from https://hugosprojects.wordpress.com/2014/03/20/implementing-aprs-gps-data/ , which is why I would prefer to send it to you in a less public way, because I'm very unsure of the licensing situation. The code produces a decodable MicE packet, which sometimes (depending on the message following the MicE) can't be decoded. Why I don't kow. Also: Be warned; my code is very much a work in progress and far from being polished. I would call it pre-alpha... ;)
MicE is a really hot mess of encodings. For a reason I don't get APRS seems to prefer weird 'ASCII'-based encodings which doesn't make a lot of sense in a slow and time constraint protocol like AFSK/AX25. If they would smartly encode the position data in binary they yould probably be able to encode all of it in the dest callsign and use the rest for error correction or whatever.

As for the FIFO-Idea: 
Yes there is FSK based APRS deployed, AFAIK mainly on 70cm. The global 2m APRS network uses a lot of the "legacy"AX.25/AFSK modulation. 
What I meant ist the following: At the moment you setup the TX in FSK mode with the FSK modulation source being GPIOx. Then you modulate the TX/FSK with the "audio"/AFSK waveform using the GPIOx (thus tone() and notone()). 
What DL7AD did with pecanpico10 ( https://github.com/DL7AD/pecanpico10 ) is to also set the TX to FSK but the modulation source being the FIFO using a carefully calculated bitrate (a multiple of the audio freq (2..4x audio fmax I guess) disabling data whitening, preambles and such. He then wrote the actual PWM ("tone") waveform to the FIFO as e.g. (in bin) 00001111 11110000 etc., getting in essence the same 'AFSK' out the RF end as when you would modulate it using the GPIO and tone. So no need for SSB/FSK demod, but the RF-signal having a higher spectral 'content' b/c the AFSK is a square-wave. 
It just lets the TRX Chip take care of the 'dirty AFSK hack' and freeing CPU ressources.

jgromes, 2021-12-25:
Thanks, I'll take a look at the test program it as soon as I'm back at my work PC.

The macro actualy expects `::attachInterrupt`, not `Module::attachInterrupt`. That's because Arduino API declares `attachInterrupt` as a function at the global scope. The signature of that global scope function is defined for each Arduino board in BuildOpt.h (since there are slight inconsistencies between different Arduino boards). 

Looking briefly at the config file, you're defining `NRF52832_XXAA`, which will cause RadioLib to assume it's compiling for Adafruit nRF52 Arduino core. That defines `int attachInterrupt()`, whereas on Platformio, it's probably `void attachInterrupt()` (haven't checked, but it explains the behavior). Unfortunately I can't change it in the BuildOpt file, that would break compatibility with Adafruit nRF52 core.

You could use `RADIOLIB_CUSTOM_ARDUINO`, which would let you to define everything by hand in the BuildOpt file. You can use the Adafruit nRF52 config as a starting point and change the signature of `attachInterrupt`.

Regarding Mic-E and AFSK - both are good ideas, I propose we move the discussion to separate threads (#430 and #431).

omegat, 2021-12-25:
OK great!
As my laptop is still almost working (burnt the display on a Christmas-candle... :expressionless:) I'll look into BuildOpt.h while defining `RADIOLIB_CUSTOM_ARDUINO`. I see what you did there, I like the implementation!!

Yeah as I said it's a very crokid build environment. The `NRF52832_XXAA` actually refers to the linkerscript. The actual board pretends to be a `PCA1000X` (defined in board json file) as this board def has straight pin definitions. I could really get mad on these arduino-wiring-style pin number confusions. 

..I've already seen the 2 other issues -> see there. :thumbsup:

jgromes, 2021-12-25:
> burnt the display on a Christmas-candle

That's unfortunate, I'm really sorry to hear that :/ 

Thanks, the idea was to allow for some way for customized builds that do not fit the predefined platforms.

> The NRF52832_XXAA actually refers to the linkerscript

It doesn't really matter, as long as its a preprocessor macro. There's actually no consistency accross Arduino boards when it comes to defining "which board is the code running on". Some specify `ARDUINO_ARCH_xxxx`, others don't bother and for those I had to find some other macro to check. Arduino framework can be surprisingly inconsistent across boards unfortunately.

omegat, 2021-12-25:
> That's unfortunate, I'm really sorry to hear that :/

Thanks! It's ~~Time~~ an Excuse to upgrade ... ;)

> It doesn't really matter, as long as its a preprocessor macro. There's actually no consistency accross Arduino boards when it comes to defining "which board is the code running on". Some specify [...]

Yeah. It's especially bad for hardware platforms which normally come in module-form like ESPx or the nRF5x. If you want to design your own hardware not using these modules you have to bend the build environments like crazy.

But I can report success: I defined `-DRADIOLIB_CUSTOM_ARDUINO` and changed the following in BuildOpt.h starting from line 63:
```
  #define RADIOLIB_CB_ARGS_PIN_MODE                   (void, pinMode, uint32_t pin, uint32_t mode)
  #define RADIOLIB_CB_ARGS_DIGITAL_WRITE              (void, digitalWrite, uint32_t pin, uint32_t value)
  #define RADIOLIB_CB_ARGS_DIGITAL_READ               (int, digitalRead, uint32_t pin)
  #define RADIOLIB_CB_ARGS_TONE                       (void, tone, uint8_t _pin, unsigned int frequency, unsigned long duration)
  #define RADIOLIB_CB_ARGS_NO_TONE                    (void, noTone, uint8_t _pin)
  #define RADIOLIB_CB_ARGS_ATTACH_INTERRUPT           (void, attachInterrupt, uint32_t interruptNum, void (*userFunc)(void), uint32_t mode)
  #define RADIOLIB_CB_ARGS_DETACH_INTERRUPT           (void, detachInterrupt, uint32_t interruptNum)
  #define RADIOLIB_CB_ARGS_YIELD                      (void, yield, void)
  #define RADIOLIB_CB_ARGS_DELAY                      (void, delay, unsigned long ms)
  #define RADIOLIB_CB_ARGS_DELAY_MICROSECONDS         (void, delayMicroseconds, uint32_t us)
  #define RADIOLIB_CB_ARGS_MILLIS                     (unsigned long, millis, void)
  #define RADIOLIB_CB_ARGS_MICROS                     (unsigned long, micros, void)
  #define RADIOLIB_CB_ARGS_SPI_BEGIN                  (void, SPIbegin, void)
  #define RADIOLIB_CB_ARGS_SPI_BEGIN_TRANSACTION      (void, SPIbeginTransaction, void)
  #define RADIOLIB_CB_ARGS_SPI_TRANSFER               (uint8_t, SPItransfer, uint8_t b)
  #define RADIOLIB_CB_ARGS_SPI_END_TRANSACTION        (void, SPIendTransaction, void)
  #define RADIOLIB_CB_ARGS_SPI_END                    (void, SPIend, void)
```
...this now seems to compile just fine. BUT it isn't complete, as there are some other "wrong" types still in there.... Whatever.
Now investigating the tone/notone issue.

jgromes, 2022-02-13:
This seems stale - @omegat feel free to reopen if there's some update.

closed 2022-02-13 by jgromes