jgromes/RadioLib #430: [APRS] Mic-E support

jgromes, 2021-12-25:
RadioLib APRS could be extended for Mic-E support as discussed in #429.

jgromes, 2021-12-25:
@omegat I think that license gives you permissions, therefore no license means no permissions. I think it will be safer to either implement from the specification, or re-use existing code under known license (which we can then reference).

omegat, 2021-12-25:
...hm. That actually is/was my concern. 
But: We could ask the guy if he is fine with the code being used in your lib to encode Mic-E. As he published it on his website I don't think he wants to keep it a secret. As he states in the about-page: 

> This blog is a place for me to document personal projects I’ve worked on in the hope that other people can learn from my mistakes. All my projects suffer from feature creep.

As there is no contact info on the blog, I posted a comment under the GPS post containing the Mic-E code.

jgromes, 2021-12-25:
Thanks, let me know if the author gets in touch. In the meantime, I'll check the specification and try to think of a nice API for this.

omegat, 2021-12-25:
Will do.
I have a ZIP containing all the code I have so far (including the code derived from the blog (mainly contained in `AprsFrame::writeMicE`) and pecanpico, which is GPLv3); also including APRS doc PDFs and my API functions (which I have no problem donating to the cause). It declares a AprsFrame class which takes care of all the funnybussines going on with APRS. 
-> How can I get that to you best if you want it?

jgromes, 2021-12-26:
I prefer to share everything as publicly as possible, not only to comply with the license, but also for personal reasons. If you have code that is yours and you're willing to contribute it to RadioLib, then a pull request is the way to go. The advantage here is that your contribution to RadioLib is logged and tracked basically forever.

If you have something that you're not sure how to implement in RadioLib (or don't want to do it yourself), you can attach it here. I'll try to integrate it and credit you in a comment, or if it can't be directly reused, I'll use your code as a base for RadioLib Mic-E implementation.

If you have something that you can't share publicly, send an email to the address in library.properties file. In this case, I may not be able to directly reuse the code, so that makes it the unpreferred option.

omegat, 2021-12-26:
[image]
https://hugosprojects.wordpress.com/2014/03/20/implementing-aprs-gps-data/

So here we go!
Find my ...I think experimental is the right word... implementation attached.
[sx127xAPRS.zip](https://github.com/jgromes/RadioLib/files/7777703/sx127xAPRS.zip)

jgromes, 2022-08-20:
Well, took a while, but Mic-E is impemeted now, using the code from @omegat - thanks!

closed 2022-08-20 by jgromes