jgromes/RadioLib #432: RaspiDuino Not supported

siravijbb, 2021-12-27:
**IMPORTANT: Before submitting an issue, please check the following:**
1. **Read [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. RadioLib has a [Wiki](https://github.com/jgromes/RadioLib/wiki) and an extensive [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there.
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/RadioLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about wireless modules, wiring, usage, etc. Only create issues for problems specific to RadioLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

**Sketch that is causing the module fail**

```c++
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  LoRa.setPins(22, 9, 8);
  while (!Serial);

  Console.println("LoRa Sender");

  if (!LoRa.begin(433E6)) {
    Console.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Console.print("Sending packet: ");
  Console.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}
```

**Hardware setup**
Raspiduino (Raspberry Pi4B which supported raspiduino

**Debug mode output**
In file included from C:\Users\sirav\Documents\Arduino\libraries\RadioLib\examples\Morse\Morse_Transmit\Morse_Transmit.ino:25:
C:\Users\sirav\Documents\Arduino\libraries\RadioLib\src/RadioLib.h:54:4: warning: #warning "RadioLib might not be compatible with this Arduino board - check supported platforms at https://github.com/jgromes/RadioLib!" [-Wcpp]
   54 |   #warning "RadioLib might not be compatible with this Arduino board - check supported platforms at https://github.com/jgromes/RadioLib!"
      |    ^~~~~~~
In file included from sketch\Morse_Transmit.ino.cpp:1:
C:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino\cores\piduino/Arduino.h:109:19: error: expected unqualified-id before 'unsigned'
  109 | #define millis() (unsigned long)(STCV / 1000)
      |                   ^~~~~~~~
C:\Users\sirav\Documents\Arduino\libraries\RadioLib\src/Module.h:329:14: note: in expansion of macro 'millis'
  329 |     uint32_t millis();
      |              ^~~~~~
C:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino\cores\piduino/Arduino.h:109:19: error: expected ')' before 'unsigned'
  109 | #define millis() (unsigned long)(STCV / 1000)
      |                  ~^~~~~~~~
C:\Users\sirav\Documents\Arduino\libraries\RadioLib\src/Module.h:329:14: note: in expansion of macro 'millis'
  329 |     uint32_t millis();
      |              ^~~~~~
C:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino\cores\piduino/Arduino.h:108:19: error: expected unqualified-id before 'unsigned'
  108 | #define micros() (unsigned long)(STCV)
      |                   ^~~~~~~~
C:\Users\sirav\Documents\Arduino\libraries\RadioLib\src/Module.h:334:14: note: in expansion of macro 'micros'
  334 |     uint32_t micros();
      |              ^~~~~~
C:\Program Files (x86)\Arduino\hardware\RaspberryPi\piduino\cores\piduino/Arduino.h:108:19: error: expected ')' before 'unsigned'
  108 | #define micros() (unsigned long)(STCV)
      |                  ~^~~~~~~~
C:\Users\sirav\Documents\Arduino\libraries\RadioLib\src/Module.h:334:14: note: in expansion of macro 'micros'
  334 |     uint32_t micros();
      |              ^~~~~~
exit status 1
Error compiling for board RaspberryPI B+/2.

**Additional info (please complete):**
 - MCU: [e.g. Arduino Uno, ESP8266 etc.]Raspberry Pi4B
 - Link to Arduino core: (https://github.com/me-no-dev/RasPiArduino)
 - Wireless module type Raspberry Pi Lan Port (upload using SSH)
 - Arduino IDE version  Lastest
 - Library version 5.0.0

jgromes, 2021-12-27:
RaspiDuino is not supported at the moment. Overall, it seems like the default platform almost worked - almost. Looks like RaspiDuino defines `millis()` and `micros()` as preprocessor macros, which seems like a strange choice and will require either fix on RaspiDuino side, or hack on RadioLib side.

Also, the posted sketch seems completely unrelated.

siravijbb, 2021-12-27:
ok,thanks .

jgromes, 2021-12-28:
Seeing as the last update to RaspiArduino was 2 years ago, I don't think we can wait for it to get fixed. I managed to hack this on RadioLib side, so the examples do compile now.

HOWEVER There's a couple warnings that originate from RasPiDuino I can't do anything about. I'm also not going to add RasPiArduino to the CI, since it's not available in Arduino Boards manager and seems to take a long time to compile. So I can't guarantee all of the examples will compile, and also didn't test on hardware. That being said, RaspiDuino should be supported now.

closed 2021-12-28 by jgromes

siravijbb, 2021-12-28:
Thank you so much for fixing it!