jgromes/RadioLib #434: Examples for Si4432 module are crashing on ESP8266

Arthurk12, 2021-12-28:
**Describe the bug**
When running [Si4432x_Receive_Interrupt.ino](https://github.com/jgromes/RadioLib/blob/master/examples/Si443x/Si443x_Receive_Interrupt/Si443x_Receive_Interrupt.ino) or [Si4432x_Transmit_Interrupt.ino](https://github.com/jgromes/RadioLib/blob/master/examples/Si443x/Si443x_Transmit_Interrupt/Si443x_Transmit_Interrupt.ino) on ESP8266 mcu, it crashes on runtime.

Error : `ISR not in IRAM!`

**To Reproduce**
1. Load the ESP8266 board with any of the above mentioned Si4432 examples
2.  Open the serial monitor
3. You shold see an error similar to the screenshot below. 

**Expected behavior**
ESP8266 mcu should not crash when running the example.

**Screenshots**
[image]

**Possible Solution**
Since ESP8266 requires the ISR functions to be stored in RAM, the ICACHE_RAM_ATTR macro could be added before the interrupt function.
Obs: I'm not shure and I don't have one to test it, but I think the same is valid for the ESP32 mcu.

Something like:
```
...
#ifdef ESP8266
  ICACHE_RAM_ATTR
#endif
void setFlag(void) {
...
```

**Additional info (please complete):**
 - MCU: ESP8266
 - Wireless module type Si4432
 - Arduino IDE version 1.8.16
 - Library version 5.0.0

jgromes, 2021-12-28:
This is the case for all interrupt examples, not just Si4432. And you're right, similar thing applies for ESP32 as well (not sure if its the same `ICACHE_RAM_ATTR` macro for both).

I was avoiding adding this to the examples because it's so platform-specific (and because I thought users would add this by themselves, since all ISRs on ESP8266/ESP32 have to placed in IRAM). However, this is not the first time it's being brought up, so it's better to add the macro to the examples and be done with it.

Arthurk12, 2021-12-29:
> This is the case for all interrupt examples, not just Si4432

I didn't know that...

> I was avoiding adding this to the examples because it's so platform-specific (and because I thought users would add this by themselves, since all ISRs on ESP8266/ESP32 have to placed in IRAM). [...]

 So, perhaps the better solution would be to add a warning in the comment above the interrupt function, saying that it might be necessary to add a platform-specific macro.
 
 Feel free to close my PR if it is not the best approach to solve the issue.

jgromes, 2021-12-29:
Turns out the same macro works on both ESP8266 as well as ESP32, so I added it to all the examples. Thanks for pointing it out.

closed 2021-12-29 by jgromes