jgromes/RadioLib #436: Code seem not to be working after using Raspiduino

siravijbb, 2021-12-29:
**IMPORTANT: Before submitting an issue, please check the following:**
1. **Read [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. RadioLib has a [Wiki](https://github.com/jgromes/RadioLib/wiki) and an extensive [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there.
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/RadioLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about wireless modules, wiring, usage, etc. Only create issues for problems specific to RadioLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

`#include <RadioLib.h> // Include Lib
#include <SPI.h>
#include <LoRa.h>
#include <TinyGPS++.h>

// Define Pins
#define LORA_SS 22
#define LORA_RST 9
#define LORA_DIO0 8
int counter = 0; // Count Time
SX1278 radio = new Module(22, 8, 9, 7);

TinyGPSPlus gps;// Create a TinyGPS++ object
MorseClient morse(&radio);

void setup() {
uint8_t syncWord[] = {0x12, 0xAD};
LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO0);  
LoRa.setSpreadingFactor (6); // Spreeding Factors
Serial.begin(9600); // connect serial
  if (!LoRa.begin(433E6)) {
    Console.println("Starting LoRa failed!");
    while (1);
  }

  
  
  LoRa.begin(433E6);// set Freq
  LoRa.setTxPower(20);//1-20 dBm
  
  
  //radio.setRfSwitchPins(X, Y);// RX enable:   X// TX enable:   Y

}

void loop() {
int state = morse.begin(433.0,10);
morse.startSignal();
morse.print("HS SRB");
state = radio.beginFSK();
state = radio.transmit("Hello World!");
Console.print ("Done ");
Console.print (counter);
counter++;
delay(5000);

}

------------------------------------------------------------ Another Code same error (example)

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(22, 8, 9, 7);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Console.print(F("[SX1278] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Console.println(F("success!"));
  } else {
    Console.print(F("failed, code "));
    Console.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */
}

void loop() {
  Console.print(F("[SX1278] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX127x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Console.println(F(" success!"));

    // print measured data rate
    Console.print(F("[SX1278] Datarate:\t"));
    Console.print(radio.getDataRate());
    Console.println(F(" bps"));

  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Console.println(F("too long!"));

  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Console.println(F("timeout!"));

  } else {
    // some other error occurred
    Console.print(F("failed, code "));
    Console.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}

`

Console Eror code 

> [SX1278] Initializing ... failed, code -2

wiring already corrected but ground with reset pin and DIO1 didn't connected since not needed

siravijbb, 2021-12-29:
sory I cant put code in code text box

jgromes, 2021-12-29:
1. I fixed the formatting for you, I suggest taking a look at some [Markdown cheat sheet](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code).

2. The first sketch is very confused - you're mixing RadioLib and LoRa library, please don't do that, things will break.

3. -2 means CHIP_NOT_FOUND, which is almost certainly bad wiring. Which leads to your remark "but ground with reset pin and DIO1 didn't connected since not needed". Do you mean you didn't connect ground? How did you arrive at the conclusion that was not needed?

siravijbb, 2021-12-29:
nono,I mean for some libary i didnt connect DIO1 and reset pin to ground.

siravijbb, 2021-12-29:
`#include <RadioLib.h> // Include Lib
#include <SPI.h>
#include <TinyGPS++.h>

int counter = 0; // Count Time
SX1278 radio = new Module(22, 8, 5, 7);

TinyGPSPlus gps;// Create a TinyGPS++ object
MorseClient morse(&radio);

void setup() {
uint8_t syncWord[] = {0x12, 0xAD};
Serial.begin(9600); // connect serial

  
  //radio.setRfSwitchPins(X, Y);// RX enable:   X// TX enable:   Y

}

void loop() {
int state = morse.begin(433.0,10);
morse.startSignal();
morse.print("HS SRB");
state = radio.beginFSK();
state = radio.transmit("Hello World!");
Console.print ("Done ");
Console.print (counter);
counter++;
delay(5000);

}`

It seem that this code work now but I don't heard the LoRa sending word ,I only hear the Mose code (SDR)

siravijbb, 2021-12-29:
```c++
#include <RadioLib.h> // Include Lib
#include <SPI.h>
#include <TinyGPS++.h>

int counter = 0; // Count Time
SX1278 radio = new Module(22, 8, 5, 7);

TinyGPSPlus gps;// Create a TinyGPS++ object
MorseClient morse(&radio);

void setup() {
  
  Serial.begin(9600); // connect serial

  //radio.setRfSwitchPins(X, Y);// RX enable:   X// TX enable:   Y

}

void loop() {
  int state = morse.begin(433.0, 10);
  morse.startSignal();
  morse.print("HS SRB");
  radio.beginFSK(433.0,48.0, 125.0,20, 16);
  uint8_t syncWord[] = {0x12, 0xAD};
  radio.transmit("Hello World!");
  Console.print ("Done ");
  Console.print (counter);
  counter++;
  delay(5000);

}
```

I changed mouse work fine,but I don't hear the FSK (LoRa) sound. what do I do wrong? I use (SDR)

siravijbb, 2021-12-29:
ah It worked now, I had to use FSK module to send mouse.

closed 2021-12-29 by siravijbb