jgromes/RadioLib #443: RADIOLIB_ERR_TX_TIMEOUT

alemori, 2022-01-17:
**IMPORTANT: Before submitting an issue, please check the following:**
1. **Read [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. RadioLib has a [Wiki](https://github.com/jgromes/RadioLib/wiki) and an extensive [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there.
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/RadioLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about wireless modules, wiring, usage, etc. Only create issues for problems specific to RadioLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

**Sketch that is causing the module fail**

```c++
paste the sketch here, even if it is an unmodified example code
```

**Hardware setup**
Wiring diagram, schematic, pictures etc.

**Debug mode output**
Enable all [debug levels](https://github.com/jgromes/RadioLib/wiki/Debug-mode) and paste the Serial monitor output here.

**Additional info (please complete):**
 - MCU: [e.g. Arduino Uno, ESP8266 etc.]
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type [e.g. CC1101, SX1268, etc.]
 - Arduino IDE version  [e.g. 1.8.5]
 - Library version [e.g. 3.0.0]

jgromes, 2022-01-17:
> still the transmission is done.

How have you verified that? By reception of the packet?

Apparent Tx timeout can be caused by bad connection on the DIO0 pin.

alemori, 2022-01-17:
Hello, I made a change of micro, because all roads said that the problem was physical, and luckily the error disappeared. Thank you very much for your prompt response.

closed 2022-01-17 by jgromes