jgromes/RadioLib #444: GPIO2 goes HIGH when calling begin for RadioLib using SX1280 chip on ESP32 DevKit v1..

f34rdotcom, 2022-01-20:
Not sure if this is a problem with Arduino/ESP32 SPI lib or something specific inside of the RadioLib. I am looking for input. I have success sending test messages with the SX1280 and ESP32 using VSPI pins but I did note a strange issue that GPIO2 on the ESP32 goes HIGH when I call begin on the Module class instances. This causes the BLUE Led on the ESP32 DevKit V1 to turn on. Any ideas what would cause this would be appreciated.

cterwilliger, 2022-01-20:
The onboard LED is connected to GPIO2 for ESP32.  Any change on this pin will be seen by the LED.  Its either a nuisance or a troubleshooting aid :).

f34rdotcom, 2022-01-21:
> The onboard LED is connected to GPIO2 for ESP32. Any change on this pin will be seen by the LED. Its either a nuisance or a troubleshooting aid :).

Exactly why I want it back off :) I use it sometimes for debugging etc. I just don't understand why and where RadioLib turned it on.

jgromes, 2022-01-21:
@f34rdotcom Could you share your code, just so that we're on the same page?

f34rdotcom, 2022-01-21:
PEBCAC.

closed 2022-01-21 by f34rdotcom