jgromes/RadioLib #456: Functions to read and write unsigned bytes

specternecter, 2022-01-28:
I'm trying to exchange data between 2 esp32s using SX1276 Lora modules and it appears RadioLib only accepts strings. I apologize if this has been answered or if I'm missing something but I have been all over the library and searched this repo many times over. I can't find any way to write the actual byte values.

Sending string data won't work for my application because there's an mcu on the other end, not a human and all I see is human-readable options. The only way I can see to make it work would require a byte for each digit of any value and that's not going to go well over Lora.

Is there a way to read and write unsigned bytes like Serial.read() and Serial.write() or Serial.readBytes() and Serial.writeBytes() do? If not, could this feature be added?

jgromes, 2022-01-28:
Of course arbitrary binary data can be transmitted, after all even string is just bytes. I think may have overlooked the examples ;)

```
// you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */
```

closed 2022-01-28 by jgromes

specternecter, 2022-01-28:
Ahhhhhh...... in fact I did. Thanks!