jgromes/RadioLib #463: NodeMCU+SX1278 don´t work

jp-polito, 2022-02-07:
**IMPORTANT: Before submitting an issue, please check the following:**
1. **Read [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. RadioLib has a [Wiki](https://github.com/jgromes/RadioLib/wiki) and an extensive [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there.
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/RadioLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about wireless modules, wiring, usage, etc. Only create issues for problems specific to RadioLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

**Sketch that is causing the module fail**

```c++
paste the sketch here, even if it is an unmodified example code
```

**Hardware setup**
Wiring diagram, schematic, pictures etc.

**Debug mode output**
Enable all [debug levels](https://github.com/jgromes/RadioLib/wiki/Debug-mode) and paste the Serial monitor output here.

**Additional info (please complete):**
 - MCU: [e.g. Arduino Uno, ESP8266 etc.]
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type [e.g. CC1101, SX1268, etc.]
 - Arduino IDE version  [e.g. 1.8.5]
 - Library version [e.g. 3.0.0]

jp-polito, 2022-02-07:
This is my code:
/*
   RadioLib AX.25 Transmit AFSK Example

   This example sends AX.25 messages using
   SX1278's FSK modem. The data is modulated
   as AFSK at 1200 baud using Bell 202 tones.

   Other modules that can be used for AX.25
   with AFSK modulation:
    - SX127x/RFM9x
    - RF69
    - SX1231
    - CC1101
    - nRF24
    - Si443x/RFM2x

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(D8, D2, D0, RADIOLIB_NC );

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

// create AFSK client instance using the FSK module
// pin 5 is connected to SX1278 DIO2
AFSKClient audio(&radio, D1);

// create AX.25 client instance using the AFSK instance
AX25Client ax25(&audio);

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.beginFSK();

  // when using one of the non-LoRa modules for AX.25
  // (RF69, CC1101,, Si4432 etc.), use the basic begin() method
  // int state = radio.begin();

  if(state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  // initialize AX.25 client
  Serial.print(F("[AX.25] Initializing ... "));
  // source station callsign:     "N7LEM"
  // source station SSID:         0
  // preamble length:             8 bytes
  state = ax25.begin("N7LEM");
  if(state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  // Sometimes, it may be required to adjust audio
  // frequencies to match the expected 1200/2200 Hz tones.
  // The following method will offset mark frequency by
  // 100 Hz up and space frequency by 100 Hz down
  /*
    Serial.print(F("[AX.25] Setting correction ... "));
    state = ax25.setCorrection(100, -100);
    if(state == RADIOLIB_ERR_NONE) {
      Serial.println(F("success!"));
    } else {
      Serial.print(F("failed, code "));
      Serial.println(state);
      while(true);
    }
  */
}

void loop() {
  // send AX.25 unnumbered information frame
  Serial.print(F("[AX.25] Sending UI frame ... "));
  // destination station callsign:     "NJ7P"
  // destination station SSID:         0
  int state = ax25.transmit("Hello World!", "NJ7P");
  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));

  } else {
    // some error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  delay(1000);
}

jp-polito, 2022-02-07:
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 01a0
3fffff80:  3fffdad0 fffffffe 3ffee680 402010d4  
3fffff90:  00000010 00000000 feefeffe feefeffe  
3fffffa0:  feefeffe 00000000 3ffee6d4 40204eac  
3fffffb0:  feefeffe feefeffe 3ffe85d8 4010126d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
⸮⸮⸮⸮⸮⸮Ԝ[SX1278] Initializing ... failed, code -2

jgromes, 2022-02-07:
> what can I do to solve this problem?

I would start by trying to figure out what does error -2 mean. Conveniently, RadioLib has a full API documentation, which includes a page about [status codes](https://jgromes.github.io/RadioLib/group__status__codes.html).

From it, you will see that -2 is `RADIOLIB_ERR_CHIP_NOT_FOUND`, and it means that RadioLib was unable to find the radio module on the pins you provided. So it's most likely a fault in the wiring, or bad pins specified.

I also don't know why are you trying to run the AX.25 AFSK example straight away - I would suggest starting by the more simple SX127x_Transmit/Receive examples first.

jp-polito, 2022-02-07:
Hi! Thanks for the answer! I found what happened very strange, I use RadioLib in arduino, and I decided to test with a nodeMCU but without success, I will follow your tips! Thanks!