jgromes/RadioLib #465: Region `FLASH' overflowed by 10476 bytes of the SX127x_Transmit demo

IOT-MCU, 2022-02-09:
Arduino: 1.8.9 (Windows 10), Board: "Generic STM32G0 series, Generic G030F6Px, STM32CubeProgrammer (SWD), Enabled (generic 'Serial'), None, Smallest (-Os default), None, Newlib Nano (default)"

Build options changed, rebuilding all
c:/users/iot/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\IOT\AppData\Local\Temp\arduino_build_773012/SX127x_Transmit.ino.elf section `.text' will not fit in region `FLASH'

c:/users/iot/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 10476 bytes

collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic STM32G0 series.

jgromes, 2022-02-09:
The error is quite self-explanatory - the compiled binary simply doesn't fit into your MCU. Unfortunately, STM32G030F6 only has 32 kB of Flash.

closed 2022-02-09 by jgromes

IOT-MCU, 2022-02-10:
Thanks, maybe STM32G030K8 or C8 is a good choice, the flash of them is 64kB