jgromes/RadioLib #466: standby();  stoped esp32

smartua7, 2022-02-09:
Hello everyone!

**Describe the bug**
`radio.transmit(""); `stops all esp32 work
I'm using the Receive_Interrupt sketch, but sending the message in blocking mode.
The sketch works as it should, but when I add my other code, the transmit function starts to hang.
This function terminates at the output of the text "Transmit in" (below), and the output of the text "standby();- OK" does not occur.
Status before executing the transmit function = 0

```
int16_t SX126x::transmit(uint8_t* data, size_t len, uint8_t addr) {
  Serial.println("transmit in"); //this is output
  int16_tstate = standby(); //stop here
  RADIOLIB_ASSERT(state);
Serial.println("standby();-OK"); //this is not displayed
```

**To reproduce**
I can't post all the code, but I can describe the behavior of the entire program.
I have VSPI for Lora SX1262 module and HSPI for display. The internal SPI is used to store the file in SPIFFS/ Since they are all separate hardware, I don't see conflict issues here.

**Expected behavior**
I expected the message to go

**Additional info (please complete):**
 - MCU: esp32-wroom
 - Wireless module type SX1262
 - Arduino IDE version 1.8.15
 -Library version=5.0.0

jgromes, 2022-02-10:
> radio.transmit(""); stops all esp32 work

Are you really transmitting an empty string? I have to admit I did not try that, so there might be some corner case that is not covered.

> I'm using the Receive_Interrupt sketch, but sending the message in blocking mode.

That's not something I would suggest doing. Why are mixing the two approaches?

> but when I add my other code, the transmit function starts to hang.

Well, then it seems to point to an issue with your "other code" (which you did not share), as opposed to an issue within RadioLib.

smartua7, 2022-02-13:
Yes, that's exactly how it is. I clarified the problem and it appeared to be in conflict with the SPI set.
For the radio module, I use the following settings:

```
SPIClass loraSpi(VSPI);
SPISettings spiSettings(1000000, MSBFIRST, SPI_MODE0);		
SX1262 radio = new Module(5, 4, 16, 17, loraSpi, spiSettings);
```

For the screen, I use these settings:

```
SPIClass einkSpi(HSPI);
DEPG0150BxS810FxX_BW epd154bw(EIRST_PIN, EIDC_PIN, EICS_PIN, EIBUSY_PIN, EICLK_PIN); // reset_pin, dc_pin, cs_pin, busy_pin, clk_pin SPI_NUM_0, FREQUENCY
```

And then I use:
loraSpi.begin();
einkSPI.begin();

And at the same time, if the module is triggered first, then the screen no longer works, and vice versa, if the screen is triggered, then everything hangs on the communication module. It turns out a problem in the mutual work of HSPI VSPI

jgromes, 2022-02-13:
I have to admit I did not try using both HSPI and VSPI at the same time on ESP32, but this is definitely an issue with the platform, so I'm closing this for now.