jgromes/RadioLib #470: SX1278 Wiring question

kookakanga, 2022-02-14:
Hello, I have a Dragino LoRa Bee (https://wiki.dragino.com/index.php?title=Lora_BEE) which is based on SX1276/SX1278. I have a question about how to connect this to an Arduino Nano. 

I have made the following pin connections as as per code comments and the link above (all digital IO)
NSS pin:   10  
DIO0 pin:  2
RESET pin: 9
DIO1 pin:  3
3.3V and Ground

However the LoRa module is still unable to be found. 

**Hardware setup**
[image]

**Sketch that is causing the module fail**
Using the example code for SX1278
```c++

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(10, 2, 9, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
```

**Debug mode output**
Enable all [debug levels](https://github.com/jgromes/RadioLib/wiki/Debug-mode) and paste the Serial monitor output here.

`12:01:55.987 -> SX127x not found! (10 of 10 tries) RADIOLIB_SX127X_REG_VERSION == 0x0000, expected 0x0012
12:01:56.078 -> No SX127x found!
12:01:56.078 -> failed, code -2
`
-2 is RADIOLIB_ERR_CHIP_NOT_FOUND

**Additional info (please complete):**
 - MCU: Arduino Nano
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type SX1278
 - Arduino IDE version  1.8.12
 - Library version 5.1.0

I have tested the chip with a raspberry pi using pylora and this works, the chip is not DOA. 
If you need any further information please let me know 
Thank you for your help

**IMPORTANT: Before submitting an issue, please check the following:**
1. **Read [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. RadioLib has a [Wiki](https://github.com/jgromes/RadioLib/wiki) and an extensive [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there.
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/RadioLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about wireless modules, wiring, usage, etc. Only create issues for problems specific to RadioLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

jgromes, 2022-02-14:
Two thoughts:

1. Did you connect the SPI? Your wiring diagram doesn't mention it.
2. Do you have some voltage level conversion? It doesn't seem like it, and since Arduino Nano is a 5V-logic device, and SX127x is a 3.3V-logic device, you risk damaging it.

kookakanga, 2022-02-14:
Thanks, the SPI was the problem 

For anyone else in the future who may be having this problem the complete connection for Arduino Nano is 
DIO0 pin: 2
DIO1 pin: 3
RESET pin: 9
NSS pin: 10
MOSI: 11
MISO: 12
SCK: 13
3.3V and Ground

Ensure you include some voltage level conversion, this is not included in this thread.
Have a good day!

closed 2022-02-14 by kookakanga