jgromes/RadioLib #471: Error compiling for board RaspberryPI B+/2.

ethr0, 2022-02-14:
I attempting to compile the Si443x_Receive_Interrupt example for Raspberry Pi but it doesn't succeed,

it compiled successfully for Arduino Uno and ESP32.

Debug mode output:
Long error like this:
/home/jak/Arduino/libraries/RadioLib-master/src/protocols/FSK4/FSK4.h:68:37: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
uint32_t _shift = 0, _shiftHz = 0;
^
/home/jak/Arduino/libraries/RadioLib-master/src/protocols/FSK4/FSK4.h:69:29: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
uint32_t _bitDuration = 0;
^
In file included from /home/jak/Arduino/libraries/RadioLib-master/src/RadioLib.h:93:0,
from /home/jak/Arduino/libraries/RadioLib-master/examples/Si443x/Si443x_Receive_Interrupt/Si443x_Receive_Interrupt.ino:18:
/home/jak/Arduino/libraries/RadioLib-master/src/protocols/APRS/APRS.h:95:20: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
char _symbol = '>';
^
/home/jak/Arduino/libraries/RadioLib-master/src/protocols/APRS/APRS.h:96:19: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
char _table = '/';
^
exit status 1
Error compiling for board RaspberryPI B+/2.

Additional info:
 - MCU: Raspberry pi
 - Arduino framework for Raspberry PI: https://github.com/me-no-dev/RasPiArduino.git
 - Arduino IDE version: 1.8.5
 - Library version: 5.1.2

jgromes, 2022-02-14:
Compiling with Arduino IDE 1.8.13 I have produces couple warnings, none of which is from the library. Seeing as the current stable release of Arduino IDE is 1.8.19, I suggest updating and trying again.

ethr0, 2022-02-17:
> Compiling with Arduino IDE 1.8.13 I have produces couple warnings, none of which is from the library. Seeing as the current stable release of Arduino IDE is 1.8.19, I suggest updating and trying again.

I tried to compile this example with 1.8.13 and 1.8.19 , nothing work , same error
another option ???

jgromes, 2022-02-18:
I'm unable to reproduce this wit the latest RaspiArduino core. Please post the full output while in verbose mode (File->Preferences->Show verbose output during: Compilation).

ethr0, 2022-02-19:
> I'm unable to reproduce this wit the latest RaspiArduino core. Please post the full output while in verbose mode (File->Preferences->Show verbose output during: Compilation).
[piduino_compiling_error.txt](https://github.com/jgromes/RadioLib/files/8101769/piduino_compiling_error.txt)

jgromes, 2022-02-19:
```
/home/jak/Arduino/libraries/RadioLib-master/src/modules/RFM2x/RFM22.h:17:29: error: expected nested-name-specifier before ‘RFM22’
 RADIOLIB_TYPE_ALIAS(Si4432, RFM22);
                             ^
/home/jak/Arduino/libraries/RadioLib-master/src/BuildOpt.h:750:63: note: in definition of macro ‘RADIOLIB_TYPE_ALIAS’
     #define RADIOLIB_TYPE_ALIAS(type, alias)            using alias = type;
                                                               ^
```

Well, those errors are completely different from what you posted previously, and seem to be pointing towards an outdated compiler. Which version of RaspiArduino are you using?

ethr0, 2022-02-19:
> ```
> /home/jak/Arduino/libraries/RadioLib-master/src/modules/RFM2x/RFM22.h:17:29: error: expected nested-name-specifier before ‘RFM22’
>  RADIOLIB_TYPE_ALIAS(Si4432, [...]

I guess: version=0.0.1
[platform.txt](https://github.com/jgromes/RadioLib/files/8102710/platform.txt)

jgromes, 2022-02-26:
I am still unable to replicate this using the latest commit from that repository. Since the platform requires the user to install GCC: what OS and GCC version are you using?

ethr0, 2022-03-03:
> 

linux opensuse

gcc --version
gcc (SUSE Linux) 11.2.1 20220103 [revision d4a1d3c4b377f1d4acb34fe1b55b5088a3f293f6]
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

jgromes, 2022-03-07:
Are you sure you're using the system gcc? You should be using the piduino toolchain, as shown here: https://github.com/me-no-dev/RasPiArduino#instructions-for-arduino-ide. Please check the version of that.

Mine is:

```
C:\SysGCC\raspberry\bin>arm-linux-gnueabihf-gcc.exe --version
arm-linux-gnueabihf-gcc.exe (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

jgromes, 2022-03-22:
Closed due to inactivity.

closed 2022-03-22 by jgromes