jgromes/RadioLib #484: [RFM96] Waiting for incoming transmission ... timeout!

suryadwiastuti, 2022-02-27:
I am using RFM96 as both receiver and sender. but I get this message when running the program "Waiting for incoming transmission ... timeout!"

this is the code i am using, is there any error i can fix?

// include the library
#include <RadioLib.h>

// RFM96 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(10, 2, 9, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//RFM96 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize RFM96 with default settings
  Serial.print(F("[RFM96] Initializing ... "));
  int state = radio.begin(434.0, 500.0, 6, 5, 0x14, 2, 20, 1);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  Serial.print(F("[RFM96] Waiting for incoming transmission ... "));

  // you can receive data as an Arduino String
  // NOTE: receive() is a blocking method!
  //       See example ReceiveInterrupt for details
  //       on non-blocking reception method.
  String str;
  int state = radio.receive(str);

  // you can also receive data as byte array
  /*
    byte byteArr[8];
    int state = radio.receive(byteArr, 8);
  */

  if (state == RADIOLIB_ERR_NONE) {
    // packet was successfully received
    Serial.println(F("success!"));

    // print the data of the packet
    Serial.print(F("[SX1278] Data:\t\t\t"));
    Serial.println(str);

    // print the RSSI (Received Signal Strength Indicator)
    // of the last received packet
    Serial.print(F("[SX1278] RSSI:\t\t\t"));
    Serial.print(radio.getRSSI());
    Serial.println(F(" dBm"));

    // print the SNR (Signal-to-Noise Ratio)
    // of the last received packet
    Serial.print(F("[SX1278] SNR:\t\t\t"));
    Serial.print(radio.getSNR());
    Serial.println(F(" dB"));

    // print frequency error
    // of the last received packet
    Serial.print(F("[SX1278] Frequency error:\t"));
    Serial.print(radio.getFrequencyError());
    Serial.println(F(" Hz"));

  } else if (state == RADIOLIB_ERR_RX_TIMEOUT) {
    // timeout occurred while waiting for a packet
    Serial.println(F("timeout!"));

  } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
    // packet was received, but is malformed
    Serial.println(F("CRC error!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }
}

suryadwiastuti, 2022-02-27:
Transmitter code

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(10, 2, 9, 3);

void setup() {
  Serial.begin(9600);
  
  Serial.print(F("[SX1278] Initializing ... "));
  // carrier frequency:           915.0 MHz
  // bandwidth:                   500.0 kHz
  // spreading factor:            6
  // coding rate:                 5
  // sync word:                   0x14
  // output power:                2 dBm
  // preamble length:             20 symbols
  // amplifier gain:              1 (maximum gain)
  int state = radio.begin(434.0, 500.0, 6, 5, 0x14, 2, 20, 1);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  Serial.print(F("[SX1278] Transmitting packet ... "));
  int state = radio.transmit("Hello World!");

  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

    // print measured data rate
    Serial.print(F("[SX1278] RSSI:\t"));
    Serial.print(radio.getRSSI());
    Serial.println(F(" dBm"));

  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(10000);
}

suryadwiastuti, 2022-02-27:
[image]

suryadwiastuti, 2022-02-27:
[image]

jgromes, 2022-02-27:
First of all, you are transmitting with spreading factor 6, meaning you have to provide the receiver with the length of data you expect to receive. This is a limitation of SF6, SF7-12 does not need it (in explicit header mode, which is the default).

Next, SF6 is quite fast - the fastest LoRa transmission can be in fact. However, you are using the blocking reception approach, which spends a lot of time not actually receiving. Since you only transmit once every 10 seconds, the chance you will receive a packet is actually very low. Interrupt-driven reception is a lot better overall, but particularly for fast transmissions.

suryadwiastuti, 2022-03-01:
hi jromes... thanks for helping me, i'm just getting started on this for my final project. In fact, there is a lot to learn.

can you tell me, is the result of this monitor correct (RSSI and SNR) why the value is mines and zero

[image]

suryadwiastuti, 2022-03-01:
hi jromes.... thanks for helping me. I just started this to finish my final project. I actually have a lot to learn, thanks for explaining :)

jgromes, 2022-03-19:
Closed due to inactivity.

closed 2022-03-19 by jgromes