jgromes/RadioLib #491: Sx1278 connection rp2040

01Space, 2022-03-14:
I use rp2040 and sx1278 to connect,
cspin -17
dio0-13
rst -7
 sck-18
 mosi-19
 miso-16
How should I modify SPI pin? Let it work normally, thank you!

jgromes, 2022-03-14:
By default, RadioLib uses the `SPI` as defined by Arduino, so whatever are the default SPI pins on RP2040, those are the ones that will be used.

01Space, 2022-03-14:
https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/RASPBERRY_PI_PICO/pins_arduino.h

// SPI
#define PIN_SPI_MISO  (16u)
#define PIN_SPI_MOSI  (19u)
#define PIN_SPI_SCK   (18u)
#define PIN_SPI_SS    (17u)
I checked raspberry in Arduino core mbed_ PI_ PICO/pins_ arduino. H documents. My connection is correct. There is no response after the firmware is brushed in, and the Lora module cannot complete initialization

jgromes, 2022-03-14:
Well then I'm missing most information I need to help you. It's all listed in the "Module not working" issue template, had you followed it.

Mainly, I need:

**Hardware setup**
Wiring diagram, schematic, pictures etc.

**Debug mode output**
Enable all [debug levels](https://github.com/jgromes/RadioLib/wiki/Debug-mode) and paste the Serial monitor output here.

01Space, 2022-03-14:
Now the data can be received successfully. After connecting the rx1278 dio1 pin to rp2040-gpio3, it works normally.
[SX1278] Data:			Hello World!
[SX1278] RSSI:			-46.00 dBm
[SX1278] SNR:			10.25 dB
[SX1278] Frequency error:	-320.86 Hz
[SX1278] Waiting for incoming transmission ... timeout!
[SX1278] Waiting for incoming transmission ... success!
[SX1278] Data:			Hello World!
[SX1278] RSSI:			-45.00 dBm
[SX1278] SNR:			11.00 dB
[SX1278] Frequency error:	-325.06 Hz
[SX1278] Waiting for incoming transmission ... timeout!
[SX1278] Waiting for incoming transmission ... timeout!
[SX1278] Waiting for incoming transmission ... success!
[SX1278] Data:			Hello World!
[SX1278] RSSI:			-45.00 dBm
[SX1278] SNR:			11.25 dB

Whether dio1 must be used or not. The information output by the serial port contains error information

[SX1278] Frequency error:	-325.06 Hz
[SX1278] Waiting for incoming transmission ... timeout!
[SX1278] Waiting for incoming transmission ... timeout!