jgromes/RadioLib #492: Is IRQ pin required to receive/transmit data?

kikolobo, 2022-03-19:
Hello,

I am using the Lora 1W - 915M30S from SparkFun.

My MCU is a teensy 4.0, however I am out of pins! And I am trying to instantiate the Module class with:

`SX1276 radio = new Module(pin_cs, RADIOLIB_NC, RADIOLIB_NC, RADIOLIB_NC);`

Whenever I do a :
`int state = radio.receive(str);`

The Teensy reboots without returning state variable, for some reason, I am guessing that the Module class is expecting an IRQ pin and that is why it is resetting, but I want to know if someone here think this is the issue so that I can look for another solution, or if someone can suggest a solution.

In the wiki I don't see that we can use RADIOLIB_NC in the IRQ. So that might as well be the issue.

Thank you for your help in advance!

jgromes, 2022-03-19:
First of all, Teensy is not officially supported, because it's not compatible (yet) with Arduino CLI which is used here for CI. I haven't tested the library on Teensy at all, so I'm actually a bit surprised it's running at all.

Next, the library currently does not support operation without IRQ, it's used to determine both the end of transmission, as well as a packet being received. So if you don't connect it, `receive()` will never return which probably triggers the watchdog, hence the reset. However, you might be able to implement it yourself using the methods `startReceive` and `readData`, combined with polling the IRQ status register, something like this:

```c++
// start continuous mode reception
radio.startReceive();

// poll the IRQ status register
while(radio.getIRQFlags() & RADIOLIB_SX127X_CLEAR_IRQ_FLAG_RX_DONE == 0);

// we should have packet now, try to read it
String str;
radio.readData(str);
```

Note that the while loop will block execution forever until a packet is received. I wrote that code without any testing, so use it at your own risk ;)

kikolobo, 2022-03-19:
Thank you so much.

It does run, finds the radio and it’s only in receiving that reboots. I think I will be better with another lib.

Is there a similar method for sending data without Irq?

jgromes, 2022-03-19:
Like I said, it's likely rebooting by watchdog only because it gets stuck in an infinite loop due to the missing IRQ pin.

I guess you could do something very similar to teh above code snippet, i.e.:

```c++
// start transmission
String str = "Hello World!";
radio.startTransmit(str);

// poll the IRQ status register
while(radio.getIRQFlags() & RADIOLIB_SX127X_CLEAR_IRQ_FLAG_TX_DONE == 0);
```

kikolobo, 2022-03-19:
I will use a temporary pin just to see if I can get it to run if you are curious if it runs ok on teensy.

jgromes, 2022-03-20:
Thanks, let me know if it works. It looks like Teensy should have Arduino CLI support at some time in the near-ish future, so then I would be able to add it as an officially supported platform here as well.

kikolobo, 2022-03-23:
Didn’t work with irq pin. Will investigate further.

jgromes, 2022-04-10:
Closed due to inactivity.

closed 2022-04-10 by jgromes