jgromes/RadioLib #494: Failing with Error -2

anyafp, 2022-03-20:
**Sketch that is causing the module fail**

```c++
/*
   RadioLib SX127x Transmit Example

   This example transmits packets using SX1278 LoRa radio module.
   Each packet contains up to 256 bytes of data, in the form of:
    - Arduino String
    - null-terminated char array (C-string)
    - arbitrary binary data (byte array)

   Other modules from SX127x/RFM9x family can also be used.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx127xrfm9x---lora-modem

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
RFM96 radio = new Module(13, 6, 14, 7);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("[RFM96] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  // some modules have an external RF switch
  // controlled via two pins (RX enable, TX enable)
  // to enable automatic control of the switch,
  // call the following method
  // RX enable:   4
  // TX enable:   5
  /*
    radio.setRfSwitchPins(4, 5);
  */
}

void loop() {
  Serial.print(F("[SX1278] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 256 characters long
  // NOTE: transmit() is a blocking method!
  //       See example SX127x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 256 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == RADIOLIB_ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

    // print measured data rate
    Serial.print(F("[SX1278] Datarate:\t"));
    Serial.print(radio.getDataRate());
    Serial.println(F(" bps"));

  } else if (state == RADIOLIB_ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F("too long!"));

  } else if (state == RADIOLIB_ERR_TX_TIMEOUT) {
    // timeout occurred while transmitting packet
    Serial.println(F("timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}
```

**Hardware setup**
[image]
[image]
[image]
[image]

**Debug mode output**
[image]

**Additional info (please complete):**
 - MCU: Arduino Uno
 - Link to Arduino core: https://github.com/arduino/ArduinoCore-avr
 - Wireless module type: RFM98
 - Arduino IDE version: 1.8.19
 - Library version: 5.1.2

**Additional questions**
I think that the pins I might be using when initializing the module might be incorrect but I'm not entirely sure how to determine which pins to use as I've been seeing different ones for different examples (and in the issues as well).

jgromes, 2022-03-20:
Your pinout indeed looks very strange. For example, you have specified pin 14 to be used as RESET pin for the radio module. I am quite confident there's no pin 14 on Arduino Uno. You have also used pin 13 for chip select - on Arduino Uno, pin 13 is SPI SCK.

SPI pins on Arduino Uno are:

SCK - 13
MISO - 12
MOSI - 11

For the other pins (NSS, DIO0 etc.), try using the default pins as specified in the example.

anyafp, 2022-03-21:
It works now! It was indeed the pinout. Thank you for the suggestion!

closed 2022-03-21 by anyafp