jgromes/RadioLib #496: Write of the SPI transfer function of SX126x.cpp checks the status.

nopnop2002, 2022-03-22:
**Describe the bug**
Hello.
I'm using sx1262.

Write of the SPI transfer function of SX126x.cpp checks the status.
https://github.com/jgromes/RadioLib/blob/master/src/modules/SX126x/SX126x.cpp#L1670

I looked at the SX1262 datasheet.

The status does not come back with SPI WRITE.****

jgromes, 2022-03-22:
> The status does not come back with SPI WRITE.

Can you point to the exact page from the datasheet that says this? As far as I can tell, status bytes is almost always returned when the microcontroller is sending data to the SX126x. Checking the status also works quite reliably in RadioLib, so I don't think there's any issue there.

[image]

nopnop2002, 2022-03-22:
When OpCode is 0x0d(=WriteRegister SPI Transaction), the status is returned, but in other cases, it is not returned.

[image]

jgromes, 2022-03-22:
Once again, I believe you are mistaken: check section 10, which describes the SPI interface in general:

[image]

But you don't have to take my word for it - feel free to enable verbose debug and run any example code for SX126x. You will see all the SPI exchanges, including status codes returned on all SPI writes. Or if you don't trust the code at all, you can always grab a scope and examine the bus ;)

nopnop2002, 2022-03-22:
I put Serial.print in SX126x :: SPItransfer.

When it is 0x86, the correct value is not returned.

```
[SX1262] Initializing ... M	SX126x
write=1 cmd=80 
A2 
write=1 cmd=80 
A2 
write=1 cmd=8F 
A2 A2 
write=1 cmd=8A 
A2 
write=1 cmd=93 
A2 
write=1 cmd=88 
A2 A2 A2 A2 A2 A2 A2 
write=1 cmd=2 
A2 A2 
write=1 cmd=8 
A2 A2 A2 A2 A2 A2 A2 A2 
write=1 cmd=89 
A2 
write=1 cmd=80 
A2 
write=0 cmd=17 
write=1 cmd=97 
A2 A2 A2 A2 
write=0 cmd=11 
Symbol length: 4.10 ms
write=1 cmd=8B 
A2 A2 A2 A2 
write=0 cmd=11 
write=1 cmd=D 7 40 
A2 A2 
write=0 cmd=11 
write=0 cmd=1D 7 36 
write=1 cmd=D 7 36 
A2 
write=1 cmd=8C 
A2 A2 A2 A2 A2 A2 
write=1 cmd=D 8 E7 
A2 
write=1 cmd=9D 
A2 
write=1 cmd=96 
A2 
setRegulatorDCDC=0
state=0
SX126x::begin=0
write=0 cmd=11 
Symbol length: 4.10 ms
write=1 cmd=8B 
A2 A2 A2 A2 
setSpreadingFactor=0
write=0 cmd=11 
Symbol length: 4.10 ms
write=1 cmd=8B 
A2 A2 A2 A2 
setBandwidth=0
write=1 cmd=98 
A2 A2 
calibrateImage=0
write=1 cmd=86 
AA 
setFrequency=-707
failed, code -707
```

jgromes, 2022-03-22:
See all the `A2`? Those are the status codes.

The actual problem happens when you attempt to calibrate frequency. You're most likely either using a device with XTAL but try to initialize TCXO (or vice versa). This also happens when you try to initialize the device for a frequency that is not supported.

Either way it's not an issue in the libary or the status code handling, but in the user level code (which you should probably post).

nopnop2002, 2022-03-22:
Thank you for responce,

>You're most likely either using a device with XTAL but try to initialize TCXO (or vice versa).

There is the SCHEMATIC DIAGRAM of the module I am using.
https://docs.ai-thinker.com/_media/lora/docs/ra-01sh_specification.pdf

There is CY3225 CRYSTAL.

----

This is the result of using another library.
The frequency is 866MHz.
The parameter for OpCode = 0x86 is 0x36, 0x20, 0x00, 0x00.
It ends correctly.

```
WriteCommand:  CMD=0x80  DataOut: 0-->A2 
WriteCommand:  CMD=0x9D  DataOut: 1-->A2 
WriteCommand:  CMD=0x89  DataOut: 7F-->A2 
WriteCommand:  CMD=0x8F  DataOut: 0-->A2 0-->A2 
WriteCommand:  CMD=0x95  DataOut: 4-->A2 7-->A2 0-->A2 1-->A2 
WriteRegister: REG=0x8E7  DataOut: 38 
WriteCommand:  CMD=0x8E  DataOut: 16-->A2 4-->A2 
WriteCommand:  CMD=0x98  DataOut: D7-->A2 D8-->A2 
WriteCommand:  CMD=0x86  DataOut: 36-->A2 20-->A2 0-->A2 0-->A2 
ReadCommand:   CMD=0xC0  DataIn: 22 
ReadCommand:   CMD=0x13  DataIn: A2 0 0 
ReadCommand:   CMD=0x14  DataIn: A2 0 0 0 
ReadCommand:   CMD=0x15  DataIn: A2 FF 
WriteCommand:  CMD=0x9F  DataOut: 0-->A2 
WriteCommand:  CMD=0xA0  DataOut: 0-->A2 
WriteCommand:  CMD=0x8A  DataOut: 1-->A2 
WriteCommand:  CMD=0x8B  DataOut: 7-->A2 4-->A2 1-->A2 0-->A2 
WriteCommand:  CMD=0x8C  DataOut: 0-->A2 8-->A2 0-->A2 FF-->A2 0-->A2 0-->A2 
WriteCommand:  CMD=0x8  DataOut: 3-->A2 FF-->A2 2-->A2 3-->A2 0-->A2 0-->A2 0-->A2 0-->A2 
SetRx timeout=16777215
WriteCommand:  CMD=0x80  DataOut: 0-->A2 
RXen:SX126x_TXEN=4 SX126x_RXEN=5
WriteCommand:  CMD=0x82  DataOut: FF-->A2 FF-->A2 FF-->A2 
ReadCommand:   CMD=0xC0  DataIn: 52 
```

----

This is the result of using your library.
The frequency is 866MHz.
The parameter for OpCode = 0x86 is 0x36, 0x20, 0x00, 0x00.
It ends incorrectly.

Is there a workaround?

```
[SX1262] Initializing ... M	SX126x
write=1 cmd=80 
0-->A2 
write=1 cmd=80 
0-->A2 
write=1 cmd=8F 
0-->A2 0-->A2 
write=1 cmd=8A 
1-->A2 
write=1 cmd=93 
20-->A2 
write=1 cmd=88 
3-->A2 16-->A2 A-->A2 0-->A2 0-->A2 0-->A2 0-->A2 
write=1 cmd=2 
3-->A2 FF-->A2 
write=1 cmd=8 
0-->A2 0-->A2 0-->A2 0-->A2 0-->A2 0-->A2 0-->A2 0-->A2 
write=1 cmd=89 
7F-->A2 
write=1 cmd=80 
0-->A2 
write=0 cmd=17 
write=1 cmd=97 
0-->A2 0-->A2 1-->A2 40-->A2 
write=0 cmd=11 
Symbol length: 4.10 ms
write=1 cmd=8B 
9-->A2 4-->A2 3-->A2 0-->A2 
write=0 cmd=11 
write=1 cmd=D 7 40 
14-->A2 24-->A2 
write=0 cmd=11 
write=0 cmd=1D 7 36 
write=1 cmd=D 7 36 
9-->A2 
write=1 cmd=8C 
0-->A2 8-->A2 0-->A2 FF-->A2 1-->A2 0-->A2 
write=1 cmd=D 8 E7 
18-->A2 
write=1 cmd=9D 
1-->A2 
write=1 cmd=96 
1-->A2 
setRegulatorDCDC=0
state=0
SX126x::begin=0
write=0 cmd=11 
Symbol length: 4.10 ms
write=1 cmd=8B 
9-->A2 4-->A2 3-->A2 0-->A2 
setSpreadingFactor=0
write=0 cmd=11 
Symbol length: 4.10 ms
write=1 cmd=8B 
9-->A2 4-->A2 3-->A2 0-->A2 
setBandwidth=0
write=1 cmd=98 
D7-->A2 DB-->A2 
calibrateImage=0
write=1 cmd=86 
36-->AA 
setFrequency=-707
failed, code -707
```

jgromes, 2022-03-22:
> There is the SCHEMATIC DIAGRAM of the module I am using.

Judging by the schematic, it is an XTAL. By default, RadioLib assumes TCXO, so you most likely left it at that. I can't really know, since you haven't posted your code.

You will have to set `tcxoVoltage = 0` in the begin method to set up the device for XTAL.

nopnop2002, 2022-03-22:
>You will have to set tcxoVoltage = 0 in the begin method to set up the device for XTAL.

I changed it like this.

```
int state = radio.begin(866.0, 125.0, 9, 7, RADIOLIB_SX126X_SYNC_WORD_PRIVATE, 10, 8, 0.0);
```

Work fine!!

Thank you for your help.

closed 2022-03-22 by nopnop2002