jgromes/RadioLib #497: RF69::setRxBandwidth floating point comparison issues

loopj, 2022-03-22:
Hey there, great lib.

I've noticed that when calling `RF69::setRxBandwidth` that the desired bandwidth doesn't actually get persisted in my applications. 

After some debugging, I discovered the culprit is the "float comparison" logic here:
https://github.com/jgromes/RadioLib/blob/c24c7750244fee68565ba6f371d056e46d1a518b/src/modules/RF69/RF69.cpp#L462-L536

Due to floating point imprecision, these comparisons always fail.

There are a few approaches I can see to fix this:
1. Use fuzzy comparison (eg. `fabs(a - b) < epsilon`
2. Convert this function to take Hz rather than kHz (a breaking change)
3. Rewrite this function to be similar to `SX127x::setRxBandwidth`, where it does a mantissa/exponent calculation instead of lookup.

I'd probably personally lean toward option 3.

Let me know if you agree and I can make a PR.

jgromes, 2022-03-22:
Thanks for pointing this out! You're right, that comparison is just plain wrong, most likely a relic of the past.

If you have a way to calculate everything using mantissa/exponent, then I would also agree that would be preferrable. If not, fuzzy comparison is fine as well (it is used in couple other places in the library).

jgromes, 2022-03-23:
Merged, thanks!

PR #498 (2022-03-23): Fix setRxBandwidth by calculating mantissa and exponent, rather than comparing floats
Used a similar approach to other modules, (CC1101/RFM95).

Tested by looping through all valid values and comparing register after setting. Also tested invalid values.

Fixes #497
closed 2022-03-23 by jgromes