jgromes/RadioLib #502: Setting bw to 500.0 and sf to 11

jgromes, 2022-03-27:
### Discussed in https://github.com/jgromes/RadioLib/discussions/495

<div type='discussions-op-text'>

<sup>Originally posted by **frogg1ven** March 21, 2022</sup>
Hi, 
I'm working with lora E220, and trying to transmitt data, but after radio.begin() I'm getting state error code -20.
My begin function has params like this: 
`radio.begin(869.5, 500.0, 11, 8, SX126X_SYNC_WORD_PRIVATE, 1, 8, 0);`
With bw=125.0 and sf=9, everything works perfectly, and I read that with bandwidth equals to 500.0 kHz, I should be able to set spreading factor to 11.
[image]
I was also trying to use `radio.setSpreadingFactor(11);` after `radio.begin(869.5, 500.0, 9, 8, SX126X_SYNC_WORD_PRIVATE, 1, 8, 0);`, and there is no state error, but it doesn't seems to change sf to 11.
Am I doing something wrong?
Regards.
</div>

jgromes, 2022-03-27:
@frogg1ven there was an issue with default BW being set to 125 kHz for all SX126x derivatives, including LLCC68. For that value, SF11 was invalid, hence the error code. Should be fixed now, thanks for reporting!

closed 2022-03-27 by jgromes

frogg1ven, 2022-03-27:
Thank you @jgromes for fast fix, everything is working fine now.
Regards.