jgromes/RadioLib #504: My CC1101 Received signal strength indicator is not working on Arduino Mega.

xxjohn10, 2022-03-28:
**IMPORTANT: Before submitting an issue, please check the following:**
1. **Read [CONTRIBUTING.md](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md)!** Issues that do not follow this document will be closed/locked/deleted/ignored.
2. RadioLib has a [Wiki](https://github.com/jgromes/RadioLib/wiki) and an extensive [API documentation](https://jgromes.github.io/RadioLib/). You might find a solution to your issue there.
3. Make sure you're using the latest release of the library! Releases can be found [here](https://github.com/jgromes/RadioLib/releases).
4. Use [Arduino forums](https://forum.arduino.cc/) to ask generic questions about wireless modules, wiring, usage, etc. Only create issues for problems specific to RadioLib!
5. Error codes, their meaning and how to fix them can be found on [this page](https://jgromes.github.io/RadioLib/group__status__codes.html).

**Sketch that is causing the module fail**
Only one change from the example, I added a delay to give the module time to get packet. 
/*
   RadioLib CC1101 Receive Example
   This example receives packets using CC1101 FSK radio module.
   To successfully receive data, the following settings have to be the same
   on both transmitter and receiver:
    - carrier frequency
    - bit rate
    - frequency deviation
    - sync word
   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#cc1101
   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// CC1101 has the following connections:
// CS pin:    10
// GDO0 pin:  2
// RST pin:   unused
// GDO2 pin:  3 (optional)
CC1101 radio = new Module(10, 2, RADIOLIB_NC, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//CC1101 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  // initialize CC1101 with default settings
  Serial.print(F("[CC1101] Initializing ... "));
  int state = radio.begin();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  Serial.print(F("[CC1101] Waiting for incoming transmission ... "));
  // you can receive data as an Arduino String
  String str;
  int state = radio.receive(str);

  // you can also receive data as byte array
  /*
    byte byteArr[8];
    int state = radio.receive(byteArr, 8);
  */
  delay(2000); // ADDED CODE
  if (state == RADIOLIB_ERR_NONE) {
    // packet was successfully received
    Serial.println(F("success!"));

    // print the data of the packet
    Serial.print(F("[CC1101] Data:\t\t"));
    Serial.println(str);

    // print RSSI (Received Signal Strength Indicator)
    // of the last received packet
    Serial.print(F("[CC1101] RSSI:\t\t"));
    Serial.print(radio.getRSSI());
    Serial.println(F(" dBm"));

    // print LQI (Link Quality Indicator)
    // of the last received packet, lower is better
    Serial.print(F("[CC1101] LQI:\t\t"));
    Serial.println(radio.getLQI());

  } else if (state == RADIOLIB_ERR_RX_TIMEOUT) {
    // timeout occurred while waiting for a packet
    Serial.println(F("timeout!"));

  } else if (state == RADIOLIB_ERR_CRC_MISMATCH) {
    // packet was received, but is malformed
    Serial.println(F("CRC error!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }
}

[image]

**Hardware setup**

I don't have a wiring diagram, but I have the pins layout

Module used: CC1101 Wireless RF Transceiver 315/433/868/915MHZ + SMA Antenna Wireless Module

Datasheet: E07-M1101D-SMA_Usermanual_EN_v1.30.pdf

CC1101 Pins -> Arduino mega pins
1 ->     PWGND                              2 ->   PW3.3V
3 ->     D02/GDO0                         4 ->   D10/CSN
5 ->     D52/SCK                             6 ->   D51/MOSI
7 ->     D50/MISO/GDO1               8 ->   D03/GDO02

**Debug mode output**
Enable all [debug levels](https://github.com/jgromes/RadioLib/wiki/Debug-mode) and paste the Serial monitor output here.

Did not use the debug mode since the module is working but one value is not outputting correctly. Module is working but the RSSI is not changing at all. Changed the distance between transmitter and receiver, no change. Tried different examples of transmit and receive, still no change. Tried a different library with CC1101 module and the RSSI worked. Is my wiring incorrect or is there another issue? The Module is getting the message from the transmitter and displaying it, but the RSSI doesn't change. LQI is also not changing but I assume that it is supposed to be at zero.

**Additional info (please complete):**
 - MCU: Arduino Mega 2560
 - Link to Arduino core: [[e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]](https://github.com/arduino/ArduinoCore-avr)
 - Wireless module type: CC1101
 - Arduino IDE version: 1.8.16
 - Library version: 5.1.2

jgromes, 2022-03-28:
> Only one change from the example, I added a delay to give the module time to get packet.

That delay does nothing. `receive()` blocks until a packet is received, or until a timeout. When `receive()` ends, you either have a packet already, or you won't receive it at all.

However, that is not your primary issue. You seem to have connected a 3.3V device directly to a 5V-logic Arduino. The CC1101 most likely aren't 5V-tolerant, so you will likely damage the module. Use some logic level converter, or a 3.3V-logic Arduino.

jgromes, 2022-03-28:
Also, this seems somewhat strange:

> D50/MISO/GDO1

Is your MISO and GDO1 tied together?

xxjohn10, 2022-03-28:
[CC1101_datasheet.pdf](https://github.com/jgromes/RadioLib/files/8366511/CC1101_datasheet.pdf)
I will step down the voltage. I believe the MISO and GDO1 is tied together based on this given datasheet. This is the data sheet for the module I am using.

jgromes, 2022-03-29:
> MISO and GDO1 is tied together based on this given datasheet

Seems to be so, though it is quite unusual I must say.

jgromes, 2022-03-30:
Most likely not, I'll try to replicate the issue you're seeing with my own hardware (even though I don't have the exact same CC1101 module).

jgromes, 2022-04-02:
I was able to replicate this issue, turns out this way due to a combination of couple minor bugs. It should be fixed now, thanks for reporting!

closed 2022-04-02 by jgromes