jgromes/RadioLib #521: Receive sensitivity from low power devices is improved by a proximate higher power source

vtjballeng, 2022-05-19:
We have multiple devices using Dorji SX1262T modules. Here is an example of the problem we're seeing.

We'll use an example with 3 devices. Device 1 & 2 are located far apart. Device 1 is broadcasting a periodic message once a second. Device 2 will not receive this broadcast message from Device 1. If we turn on Device 3, which also sends a periodic broadcast message, then Device 2 will receive the message from Device 3 and also begin to receive messages from Device 1. 

FREQUENCY_CENTER_TRACK_STATE, 915, 
LORA_SF7, 
LORA_BW_500, 
LORA_CR_4_5, 
LDRO_AUTO

Messages are ~35bytes and we see no DIO1 pin activity when Device 3 is off but Device 1 is on at Device 2. All devices have Rx boosted gain active.

jgromes, 2022-05-22:
Do you have a record of the RSSI and SNR values? Especially the SNR would be quite interesting to know, since the 500 kHz/SF7 combination is quite fast (as LoRa goes), but the required SNR on the receiving end is somewhat higher because of it. If the SNR is at the limit (or close to it), then the demodulator might have trouble locking onto the preamble. Maybe it could be worthwile to try transmitting with SF8 (which works in lower SNR than SF7), at teh same distance, transmitted power, etc., to see if the problem persists.

vtjballeng, 2022-05-23:
@jgromes , we have some recordings but we might run the simulation again to check for SNR specifically. I'll say though that we aren't talking about a general reception problem. 

This is a mode we're seeing the transceivers fall into after some period of time where they stop seeing distant devices they could previously see. Like they've dropped some receive sensitivity. We can then poke them back to full sensitivity again with a high power proximate source message and they suddenly begin to see the distant device messages again.

With respect to the problem we're seeing, why would a higher power proximate source (device 3) message improve the preamble locking for the device that is further away (device 1)?

jgromes, 2022-05-23:
Higher power would help, but that would increase both RSSI and SNR. Whereas by using a higher SF, the signal strength at the receiver would remain the same, and only the SNR limit would be increased.

That's why it would be interesting to see the SNR/RSSI values throughout the time when it issue appears - it would be interesting to see if the values before and after this issue appears are different.

jgromes, 2022-06-25:
Possibly related: https://forum.lora-developers.semtech.com/t/sx1268-problem-in-rx-continuous-mode/1122

klemzi, 2023-04-11:
> Possibly related: https://forum.lora-developers.semtech.com/t/sx1268-problem-in-rx-continuous-mode/1122

Hi @jgromes, I am having similar issue with this using this library. We were using the heltec v2 which is using the SX1276 chip and we had no issue. Now it is phased out and we decided to try out the new v3 (uses SX1262) using RadioLib. However trying out the example code for transmitting and receiving, found out that the receiver is dropping or losing a lot of packets. I need help, thanks!

jgromes, 2023-04-11:
@klemzi I don't think this is the same problem. Your signal is very strong, RSSI -19 dBm, SNR > 8 dB suggests you have the devices right next to each other transmitting at high power. The original issue was for a devices at or near the sensitivity edge. Please open a new issue (and maybe try the Receive_Interrupt examples instead of the blocking ones)with all the information requested by the appropriate template.

jgromes, 2026-03-01:
Sicne this issue is ancient at this point, and there has been no progress on it, I will go ahead and close it. For the record, I have started an errata page on the wiki which includes this problem. The only currently kjnown workaround is to periodically restart reception.

closed 2026-03-01 by jgromes

GUVWAF, 2026-03-01:
FYI, testing in the Meshtastic community revealed that just restarting reception does not (fully?) fix it, while including a re-calibration procedure does: https://github.com/meshtastic/firmware/pull/9705.

weebl2000, 2026-03-01:
> Yes, I agree it would be a nice generic function. The work has been done by [@weebl2000](https://github.com/weebl2000), hence the question to them whether they would like to upstream it? Else I can give it a go.

I'll have a look.