jgromes/RadioLib #522: CubeCell Dev Board, FSK mode: Initialization failed, code -20 (RADIOLIB_ERR_WRONG_MODEM)

ssriblo, 2022-05-22:
Hello Jan!
sorry for dumb question, but I carefully read all corresponded issues regarding  CubeCell. FSK modem works great at SX1276+ESP32

But my problem is CubeCell at FSK mode. Can not pass initialization step, failed at *radio.beginFSK();*:
```
[SX1262] Initializing ... failed, code -20
```
Code get from library example here: https://github.com/jgromes/RadioLib/blob/master/examples/SX126x/SX126x_FSK_Modem/SX126x_FSK_Modem.ino

*Sketch:*
```
#include <RadioLib.h>
// SX1262 radio = new Module(10, 2, 3, 9);
// or using CubeCell
SX1262 radio = new Module(RADIOLIB_BUILTIN_MODULE);

void setup() {
  Serial.begin(115200);
  // initialize SX1262 FSK modem with default settings
  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.beginFSK();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
```

*setup*
 - CubeCell Dev Board HTCC-AB01
 - Arduino IDE version 1.8.9
 - Library version 5.1.2

Additionally, if try to init to LoRa it passed well. But if try radio.begin(); and then sequentially radio.beginFSK(); - error -20 or -102 (RADIOLIB_ERR_INVALID_FREQUENCY_DEVIATION) occurred.

jgromes, 2022-05-22:
Could you try running the sketch in [debug mode](https://github.com/jgromes/RadioLib/wiki/Debug-mode)? That might provide a bit more insight as to what exactly is failing.

ssriblo, 2022-05-23:
Hi Jan, appreciate your response, debug log:
```
Copyright @2019-2020 Heltec Automation.All rights reserved.
[SX1278] Initializing ... M	SX126x
CMD	80	
DATW	0	AA	

CMD	80	
DATW	0	A2	

CMD	80	
DATW	0	A2	

CMD	8F	
DATW	0	A2	0	A2	

CMD	8A	
DATW	0	A2	

CMD	93	
DATW	20	A2	

CMD	88	
DATW	3	A2	D	A2	A	A2	0	A2	0	A2	0	A2	0	A2	

CMD	2	
DATW	3	A2	FF	A2	

CMD	8	
DATW	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	

CMD	89	
DATW	7F	A2	

CMD	11	
DATR	0	AA	0	FF	

failed, code -20
```

ssriblo, 2022-05-23:
as far as I see at log, failed at this code chunk (/RadioLib/src/modules/SX126x/SX126x.cpp line #107):
```
  // set TCXO control, if requested
  if(tcxoVoltage > 0.0) {
    state = setTCXO(tcxoVoltage);
    RADIOLIB_ASSERT(state);
  }

  // configure publicly accessible settings
  state = setBitRate(br);  // <-- checkink for if(getPacketType() != RADIOLIB_SX126X_PACKET_TYPE_GFSK) failed here 
  RADIOLIB_ASSERT(state);
```
But I've not idea why it happened

jgromes, 2022-05-23:
Actually, the issue seems to have happened at this point:

```
CMD	80	
DATW	0	AA	
```

The 0xAA status is a "failure to execute command". Since the previous command was calibration, I would guess the device failed to calibrate. Are you sure the device type (SX1261 vs SX1262 vs SX1268) and the frequency is correct? `beginFSK` without arguments assumes 434 MHz by default.

ssriblo, 2022-05-23:
Hi Jan,
thanks a lot! You got me brilant point!
I never saw before that Rx Bw need setup preciously like this:
} else if(fabs(rxBw - 234.3) <= 0.001) {
_rxBw = RADIOLIB_SX126X_GFSK_RX_BW_234_3;

I must to study datasheet better

Thank you @jgromes <https://github.com/jgromes> for fast reply, everything
is working fine now.

Regards,
Sergey

jgromes, 2022-05-23:
in general yes, Rx bandwidth has to be set exactly. However, if that's not the case, a "wrong Rx BW" error code should be returned instead of the "wrong mode one". Are you calling `beginFSK` without arguments? And using which module, SX1262 or SX1268?

ssriblo, 2022-05-23:
Now use following:
  int state = radio.beginFSK(868.0, 10.0, 5.0, 234.3, 10, 16, 1.6, false);
I use CubeCell Dev Board SX1262 at ASR6501 SoC

After that init procedure passed. But packets received are very rare,
insatiable.

BTW, my other FSK setup with ESP32/SX1276 works fine. Looks like I need to
examine more SX1262 parameters.

jgromes, 2022-05-28:
@ssriblo I found an issue in the default configuration of frequency deviation for SX1262, so that may have been related. I tried replicating the issues you're seeing with SX1262 and an Arduino Uno (as I don't have a CubeCell), unfortunately I was not able to replicate the issue.

Also, the 234 kHz Rx bandwith you're using seems somewhat high for the 10 kHz signal. I would suggest starting with the default settings, and going from there.

closed 2022-05-28 by jgromes